xioTechnologies/Fusion #2: Is there any dependency relative to accel&gyro ODR and filter update rate?

Alvedro, 2022-04-05:
In some places when there is talk about the implementation of the Madgwick filter often I read that the update rate of accel&gyro data from IMU should be about 5-10 smaller than the filter update rate. Based on it when I set up IMU to get accel&gyro with a 50 Hz rate, then the filter should work with a 250 Hz - 500 Hz rate. Is it true?

xioTechnologies, 2022-04-05:
The original 2011 Madgwick algorithm is very different to Fusion.  This recommendation would not be applicable.  You should call the update function at the gyroscope sample rate.  You should also configure a sample rate appropriate for the application.  50 Hz is very low.

If you are unsure how the sample rate affects accuracy then I suggest you log data from your application at the highest available sample rate, and then compare the Fusion output for this sample rate with [downsampled](https://en.wikipedia.org/wiki/Downsampling_(signal_processing)) versions of the same data.  You will likely see that sample rates <100 Hz result in significant errors, and sample rates >400 Hz provide diminishing returns.

Alvedro, 2022-04-05:
I will test, thank you for your answer.

closed 2022-04-06 by Alvedro