xioTechnologies/Fusion #3: Can't import imufusion

IKetchup, 2022-04-11:
hello,

I would lie to install imufusion however I cant install it. I get the following error with pip installation
```
ERROR: Could not find a version that satisfies the requirement imufusion==1.0.0 (from versions: none)
ERROR: No matching distribution found for imufusion==1.0.0

```
Did you already encounter this problem or do you have an idea of how to fix it?

maynardephd, 2022-04-13:
More specifically, the error is:

RuntimeError: module compiled against API version 0xf but this version of numpy is 0xe

I've upgraded to numpy 1.22.3 but I still get this error.

xioTechnologies, 2022-04-13:
The [CI](https://github.com/xioTechnologies/Fusion/actions) confirms that the Python package installs and runs for all tested OSs and Python versions.

If you are having issues installing the package then please update Python and NumPy to the latest versions.  If you have multiple versions installed then you must make sure you are running the intended version.  If you are still unable to install the Python package then please share all of the following so that we can look into it.

- Error message
- OS (inc. version)
- Python version
- NumPy version

IKetchup, 2022-04-14:
@maynardephd I managed to install imufusion.

I created a new conda environement with python 3.9 (on windows 10)
- Install numpy 1.22.3
- install imufusion

closed 2022-04-14 by IKetchup

maynardephd, 2022-04-14:
Error message: RuntimeError: module compiled against API version 0xf but this version of numpy is 0xe
I am running on Windows 10
Python version 3.10.2 using the WinPython distro
numpy 1.22.3

I will try an anaconda install and see how that does.

limited-fion, 2023-02-10:
I couldn't install the python package too.

Error message:
```
pip install imufusion
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement imufusion
ERROR: No matching distribution found for imufusion
```
OS Version:
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
Python version: 3.9.2
Numpy version: 1.24.2