xioTechnologies/Fusion #7: No matching distribution found for imufusion, Arm, Raspberry Pi

DandrewsDev, 2022-04-16:
I would love to use this for a project Im working on, but does not seem like there is any support for Pi boards right now.

When I try to run "pip install imufusion" it errors out 
ERROR: Could not find a version that satisfies the requirement imufusion
ERROR: No matching distribution found for imufusion

After digging adding the -v flag and digging into the output it appears to be because there is currently no Arm support. Is something like that possible?

xioTechnologies, 2022-04-16:
The Python package installed using `pip install` is limited to the platforms supported by [GitHub Actions Runner](https://github.com/actions/runner).  If your platform is not supported then I suggest you instead run [setup.py](https://github.com/xioTechnologies/Fusion/blob/main/Python/Python-C-API/setup.py) to build and install the package on your target platform.

DandrewsDev, 2022-04-16:
Was able to build and use the package locally. Adding the steps I needed to follow below incase anyone else ever runs into this.

Ran into errors using numpy from pip on raspberry pi. 
Thats fixable following the steps here https://numpy.org/devdocs/user/troubleshooting-importerror.html
1. pip3 uninstall numpy  # remove previously installed version
2. sudo apt install python3-numpy

1. Checkout clone code
2. cd Fusion/Python/Python-C-API
3. python3 setup.py
4. cd dist
5. unzip imufusion-1.0.1-py3.9-linux-armv7l.egg
6. cp imufusion.cpython-39-arm-linux-gnueabihf.so and imufusion.py to project libraries
7. import numpy, imufusion

And now its working great. Thanks for the info.

closed 2022-04-16 by DandrewsDev

bshmid, 2022-06-02:
> Why are you rearranging the files? You should download the whole repository and run setup.py as it is.
Thank you very much!
I just did not know it was there.