xioTechnologies/Fusion #8: GPS Example

Julspower, 2022-04-19:
Hi
the readme state its compatible with system with gps correction, it seem like the solution for lets say a fixed wing model airplane if its in a bank long enough you need a gps to make the fusion work correctly
but I do no see example using gps data, nor gps input function the the code.
can you help me there?

xioTechnologies, 2022-04-19:
The [README](https://github.com/xioTechnologies/Fusion/blob/main/README.md) states that the the algorithm can be used for _"systems that use a gyroscope and accelerometer combined with an external source of heading measurement such as GPS"_.  In this statement, GPS is only suggested an example of an external source of heading measurement.  The algorithm does not support integration of GPS data in any other way.

The algorithm includes a linear acceleration rejection feature.  This feature will prevent the centripetal acceleration associated with the banks of a fixed wing model airplane from corrupting the orientation measurement, provided that the period of the bank does not exceed the `rejectionTimeout` period configured in the algorithm settings.

Julspower, 2022-04-19:
Thank you for the clarification.
do you intend to add gps integration one day in the future?

xioTechnologies, 2022-04-19:
Not at this time.