xioTechnologies/Fusion #11: Python how to read quaternion

sierra118, 2022-04-22:
Hi, thanks for your work.
I'm curently using imufusion in order to determine the orientation of differents IMU, but i don't find how to read the quaternion before the convertion to euler  angle. How can i directly read the quaternion output ?

DocSunset, 2022-04-22:
On Python/Python-C-API/Quaternion.h around line 82 you can see get/set definitions for the quaternion python object. So I believe if you have an ahrs object `ahrs`, you could read the quaternion as `ahrs.quaternion.array` or `ahrs.quaternion.w; ahrs.quaternion.x; # etc`

It would be nice to have more approachable documentation besides the source code, but the Python-C-API is reasonably readable, so I can understand why the maintainer(s) might choose to prioritize other things.

sierra118, 2022-04-22:
Thanks for your answer it works

closed 2022-04-22 by sierra118