xioTechnologies/Fusion #9: Why FusionAhrsGetQuaternion() gets conjugated quaternion?

JerrickRowe, 2022-04-22:
```
/**
 * @brief Returns the quaternion describing the sensor relative to the Earth.
 * @param ahrs AHRS algorithm structure.
 * @return Quaternion describing the sensor relative to the Earth.
 */
FusionQuaternion FusionAhrsGetQuaternion(const FusionAhrs *const ahrs) {
    return FusionQuaternionConjugate(ahrs->quaternion);
}
```
I tried to compair Fusion with original MadgewickAHRS, and found that Fusion gives reversed quaternion result. It appears that quaternion get by `FusionAhrsGetQuaternion()` always be conjugated, thus result in reversed quaternion orientation. But if i change it to `return ahrs->quaternion;` then `FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));` returns reversed euler angle.

According to https://quaternions.online/, convert quaternion wxyz{0.96,-0.009,-0.28,-0.024} to euler angle should be xyz{-2.085,-32.467,-3.471}. But `FusionQuaternionToEuler((FusionQuaternion){0.96,-0.009,-0.28,-0.024});` returns {2.084,32.491,3.469}, which every axis are reversed. So I think something goes wrong in conversion between quaternion to euler angle.

The problem can be fixed by both applying the method from Wikipedia or just simply reverse signs in the original formula.

### Original 
```
static inline FusionEuler FusionQuaternionToEuler(const FusionQuaternion quaternion) {
#define Q quaternion.element
    const float qwqwMinusHalf = Q.w * Q.w - 0.5f; // calculate common terms to avoid repeated operations
    FusionEuler euler;
    euler.angle.roll = FusionRadiansToDegrees(atan2f(Q.y * Q.z - Q.w * Q.x, qwqwMinusHalf + Q.z * Q.z));
    euler.angle.pitch = FusionRadiansToDegrees(-1.0f * asinf(2.0f * (Q.x * Q.z + Q.w * Q.y)));
    euler.angle.yaw = FusionRadiansToDegrees(atan2f(Q.x * Q.y - Q.w * Q.z, qwqwMinusHalf + Q.x * Q.x));
    return euler;
#undef Q
}
```
### Sign reversed
```
// Sign reversed
static inline FusionEuler FusionQuaternionToEuler(const FusionQuaternion quaternion) {
#define Q quaternion.element
    const float qwqwMinusHalf = Q.w * Q.w - 0.5f; // calculate common terms to avoid repeated operations
    FusionEuler euler;
    euler.angle.roll = FusionRadiansToDegrees(-1.0f * atan2f(Q.y * Q.z - Q.w * Q.x, qwqwMinusHalf + Q.z * Q.z));
    euler.angle.pitch = FusionRadiansToDegrees(asinf(2.0f * (Q.x * Q.z + Q.w * Q.y)));
    euler.angle.yaw = FusionRadiansToDegrees(-1.0f * atan2f(Q.x * Q.y - Q.w * Q.z, qwqwMinusHalf + Q.x * Q.x));
    return euler;
#undef Q
}
```
### Method from Wikipedia
```
// Source:https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles
static inline FusionEuler FusionQuaternionToEuler(const FusionQuaternion quaternion) {
#define Q quaternion.element
    FusionEuler euler;
    // Source:https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles
    // roll (x-axis rotation)
    float sinr_cosp = 2 * (Q.w * Q.x + Q.y * Q.z);
    float cosr_cosp = 1 - 2 * (Q.x * Q.x + Q.y * Q.y);
    euler.angle.roll = FusionRadiansToDegrees(atan2f(sinr_cosp, cosr_cosp));

    // pitch (y-axis rotation)
    float sinp = 2 * (Q.w * Q.y - Q.z * Q.x);
    if (fabsf(sinp) >= 1)
        euler.angle.pitch = FusionRadiansToDegrees(copysignf(M_PI / 2, sinp)); // use 90 degrees if out of range
    else
        euler.angle.pitch = FusionRadiansToDegrees(asinf(sinp));

    // yaw (z-axis rotation)
    float siny_cosp = 2.0f * (Q.w * Q.z + Q.x * Q.y);
    float cosy_cosp = 1.0f - 2.0f * (Q.y * Q.y + Q.z * Q.z);
    euler.angle.yaw = FusionRadiansToDegrees(atan2f(siny_cosp, cosy_cosp));
    return euler;
#undef Q
}
```
I currently use the 'Sign reversed' version, and a pull request will be sent, tell me if I miss something.

DocSunset, 2022-04-22:
I also noticed that when exploring the library recently that I had to invert my gyroscope measurements to get the expected results, which seems related. I assumed that this meant the library must be expecting left-handed measurements...

JerrickRowe, 2022-04-22:
I think that's a problem not about handedness of measurement, but about handedness consistancy. Consistancy should be maintained within the library. That say, left-handed quaternion should be converted into left-handed euler angle, not right-handed. Actually, I have tried to input reversed data into Fusion, but still get reversed orientation between quaternion and the converted euler angle.

xioTechnologies, 2022-04-25:
Thank you for raising this issue.  The problem has nothing to do with handedness, and reversing the signs of the Euler angles is not a solution.

The problem is due to an inconsistency in the frame of reference assumed by `FusionQuaternionToMatrix` and `FusionQuaternionToEuler`, i,e, if the quaternion describes the sensor relative to the Earth, or the Earth relative to the sensor.

## Investigation

I found contradictions between sources on what the correct conversion of a quaternion to a rotation matrix, or a quaternion to ZYX Euler angles should be.  To investigate this, I generated a sequence of quaternions and calculated the converted values according to each source.  I then compared the values of each source with every other source.  The code used to obtain the results is available [here](https://github.com/xioTechnologies/Quaternion-Conversion-Tests).

Sources:

1.  [euclideanspace.com](https://www.euclideanspace.com/maths/geometry/rotations/conversions/)
2. [Fusion v1.0.1](https://github.com/xioTechnologies/Fusion/releases/tag/v1.0.1)
3. [Quaternions and Rotation Sequence by Jack B. Kuipers](https://www.google.co.uk/books/edition/Quaternions_and_Rotation_Sequences/p_3RDwAAQBAJ) (page 168)
4. MATLAB ([`quat2eul`](https://www.mathworks.com/help/robotics/ref/quat2eul.html), [`quat2rotm`](https://www.mathworks.com/help/robotics/ref/quat2rotm.html))
5. [scipy](https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.html)
6. [wikipedia](https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles)

### Quaternion to rotation matrix

|                | e'space  | Fusion  | Kuipers  | MATLAB  | scipy  | wikipedia  |
|----------------|----------|---------|----------|---------|--------|------------|
| **e'space**    | -        | ✗       | ✗        | ✓       | ✓      | ✓          |
| **Fusion**     | ✗        | -       | ✓        | ✗       | ✗      | ✗          |
| **Kuipers**    | ✗        | ✓       | -        | ✗       | ✗      | ✗          |
| **MATLAB**     | ✓        | ✗       | ✗        | -       | ✓      | ✓          |
| **scipy**      | ✓        | ✗       | ✗        | ✓       | -      | ✓          |
| **wikipedia**  | ✓        | ✗       | ✗        | ✓       | ✓      | -          |

✓ = Rotation matrices match, ✗ = Rotation matrices different

All sources except Fusion and Kuipers provide matching results.  While Fusion and Kuipers do not match any other sources, they do match each other.  This is because Kuipers was used as a reference when writing Fusion.  **If the Fusion and Kuipers rotation matrices are transposed then then all sources provide matching results**.  It is not clear why Kuipers uses an inverted frame of reference for rotation matrices.  I have gone through the maths in the book and it is internally consistent, it just adopts an apparently unique convention.

### Quaternion to ZYX Euler angles

|                | e'space  | Fusion  | Kuipers  | MATLAB  | scipy   | wikipedia |
|----------------|----------|---------|----------|---------|---------|-----------|
| **e'space**    | -        | ✗ (104) | ✗ (96)   | ✗ (96)  | ✗ (72)  | ✗ (96)    |
| **Fusion**     | ✗ (104)  | -       | ✗ (104)  | ✗ (104) | ✗ (104) | ✗ (104)   |
| **Kuipers**    | ✗ (96)   | ✗ (104) | -        | ✓       | ✗ (80)  | ✓         |
| **MATLAB**     | ✗ (96)   | ✗ (104) | ✓        | -       | ✗ (80)  | ✓         |
| **scipy**      | ✗ (72)   | ✗ (104) | ✗ (80)   | ✗ (80)  | -       | ✗ (80)    |
| **wikipedia**  | ✗ (96)   | ✗ (104) | ✓        | ✓       | ✗ (80)  | -         |

✓ = Euler angles match, ✗ (n) = n of 105 Euler angles different

Kuipers, MATLAB, and wikipedia provide matching results.  **If Fusion were to convert the quaternion conjugate then it would match Kuipers, MATLAB, and wikipedia.**  I manually entered a few quaternions into [quaternions.online](https://quaternions.online/) and it appears to also be consistent with Kuipers, MATLAB, and wikipedia.  The euclideanspace results contain many differences because of a [deliberate deviation in convention](https://www.euclideanspace.com/maths/geometry/rotations/conversions/quaternionToEuler/index.htm).  It is not clear why scipy provides some different results, I examined specific numerical instances but an explanation was not immediately apparent.

## Description of the problem

`FusionQuaternionToMatrix` was written using Kuipers and so includes an unintentional inversion of the frame of reference, equivalent to a quaternion conjugation or matrix transpose.  When inconsistencies were discovered during development, the Kuipers conversion was assumed to be correct and other calculations adjusted accordingly.  This resulted in `FusionQuaternionToEuler` and `FusionAhrsGetQuaternion` incorporating a quaternion conjugation to correct for the Kuipers inversion.

## Solution

- `FusionQuaternionToMatrix` will be updated to remove the unintentional inversion.
- `FusionQuaternionToEuler` will be updated to provide results consistent with Kuipers, MATLAB, and wikipedia.
- `FusionAhrsGetQuaternion` will be updated to remove the quaternion conjugation.

### Backward compatibility

System that use `FusionAhrsGetQuaternion` only in conjunction with `FusionQuaternionToMatrix` or `FusionQuaternionToEuler` will not be affected by the above changes.  Any other use of `FusionAhrsGetQuaternion` must be updated to use the quaternion conjugate once the above changes have been introduced.

xioTechnologies, 2022-04-25:
The solution has been merged as commit [92335b1](https://github.com/xioTechnologies/Fusion/commit/92335b1faaa4630c739f10a1f8c94fbf6ba33055).

closed 2022-04-25 by xioTechnologies