xioTechnologies/Fusion #16: What format is input gyroscope and accel?

David-OConnor, 2022-05-13:
```c
 * @param gyroscope Gyroscope measurement in degrees per second.
void FusionAhrsUpdate(FusionAhrs *const ahrs, const FusionVector gyroscope, const FusionVector accelerometer, const // ...
```

I have three guesses:
- Pitch, roll, yaw
- Roll, pitch, yaw
- Normalized 3d vector, ie requires processing of raw gyro measurements into a vector. (Most likely?)

Re accel: I'm assuming it's x, y, z. Is that correct?

Thank you!

xioTechnologies, 2022-05-15:
The type of the argument `gyroscope` is `FusionVector`.  This is a structure with three members: `x`, `y`, and `z`.  These are the x, y, and z axis measurements provided by the gyroscope.

Pitch, roll, and yaw are Euler **angles**.  These terms are not applicable to gyroscope measurements of **angular velocity**.  The word "normalized" has a [specific meaning](https://en.wikipedia.org/wiki/Unit_vector) when referring to vectors.  The gyroscope vector should not be normalised.

David-OConnor, 2022-05-15:
Thank you for the detailed explanation!

The IMU gyroscopes I've dealt with measure angular velocity in 3 axes, which can be described a pitch, roll, and yaw. How would you use these in this library's update function API? Is it one of the 3 bullets I posted above? Thank you.

xioTechnologies, 2022-05-15:
Gyroscope axes are usually described as: x, y, and z.  It sounds like the gyroscope you are using has instead assigned the labels: pitch, roll, and yaw.  The correspondence of these labels is ambiguous.  Are you able to share a link to the documentation?

David-OConnor, 2022-05-15:
Looking at the datasheet, it's described as you say - around the X, Y, and Z axes. The pitch/yaw/roll terminology isn't part of the IMU, but a convention I've been using, where x, y, and z are linear directions, and pitch, roll, and yaw are angular directions. With this in mind, your explanation solves the issue entirely.

[image]

So, the mapping I'm now using, from a an internal struct works out to be:
```rust
let mut gyro_data_for_ahrs = Vec3 {
    x: imu_data.v_pitch,
    y: imu_data.v_roll,
    z: imu_data.v_yaw,
};
let mut accel_data_for_ahrs = Vec3 {
    x: imu_data.a_x,
    y: imu_data.a_y,
    z: imu_data.a_z,
};
```

closed 2022-05-18 by David-OConnor

David-OConnor, 2023-05-22:
Fixed by https://github.com/xioTechnologies/Fusion/issues/9