xioTechnologies/Fusion #18: Algorithm problems

viatto, 2022-05-23:
Hi, I tried to use this algorithm to obtain the acceleration in the global reference frame by multiplying accelerometer data for the rotation matrix obtained by the Madgwick filter. When I move the object in the direction of one axis, I notice that the acceleration rises in all the axis almost equally.  What can I do to solve the problem?

xioTechnologies, 2022-05-23:
Acceleration in the global reference frame is provided by the function `FusionAhrsGetEarthAcceleration`.  I suggest you use this function instead of implementing the calculation yourself.

viatto, 2022-05-24:
Thank you, I will try! I am using the python library, Do i have to use the same input and output of the Matlab library?

xioTechnologies, 2022-05-24:
No. Fusion is only available for C/C++ and Python.