xioTechnologies/Fusion #24: Unstable yaw data

viatto, 2022-05-30:
Hi, I am sorry to bother you again, I am looking to have accurate results in the acceleration data but I found out that the yaw measure is very unstable. I tried to test the device by slowly rotating it to 45 degrees each step and here are the results.
[image]
The rotations of 45 degrees are 3 and have been performed with the same orientation.
I calibrate the magnetometer based on LS ellipsoid fitting of 15000 samples and for the gyroscope and accelerometer just the offset in static conditions obtained by mediating 500 samples. Therefore, I don't think can influence the algorithm.
Can you help me to understand what could be the problem?

viatto, 2022-05-30:
I read in the documentation that the magnetometer rejection should be at 20, now I have set it to 0.
The result I have obtained is the following:
[image]
Seems better but still strange what do you think?

xioTechnologies, 2022-05-30:
It looks like you have assigned sensor axes incorrectly. What country are you in?  The magnetometer measurements combined with the pitch and roll  indicate a magnetic inclination of around -60 degrees which would put you in either South Africa or Southern Australia.  I think the more likely explanation is that your magnetometer is upside-down and you need to fix the axes assignments.

[image]

[The US/UK World Magnetic Model for 2020-2025](http://www.geomag.bgs.ac.uk/documents/WMM2020_Report.pdf)

viatto, 2022-05-30:
I am in Belgium so yes you are right should be 60 degrees and not -60 degree. How Can I fix the axis? I shoul put the magnetometer down?
[image]
This the position of my magnetometer, the z axis is pointing toward the ground

xioTechnologies, 2022-05-30:
You need to decide what your intended XYZ axis directions are, and then use the sensor measurements that correspond to these directions.

xioTechnologies, 2022-05-30:
Your appear to be using the [MPU-9250](https://invensense.tdk.com/products/motion-tracking/9-axis/mpu-9250/).  The device datasheet confirms that the magnetometer axes are not aligned with the gyroscope and accelerometer axes.  The magnetometer is effectively upside-down relative to the gyroscope and accelerometer.

[image]

You must therefore use the following axis assignments when passing the magnetometer values.

`magnetometer = numpy.array([mpu9250_mag_y,  mpu9250_mag_x, -1 * mpu9250_mag_z])`

viatto, 2022-06-01:
Yes as you can see from the foto of the datasheet y should be switched with x, also the measurements are normalized to 1 because as @xioTechnologies mention it doesn't matter for attitude the magnitude of the measurement

closed 2022-06-07 by xioTechnologies