xioTechnologies/Fusion #25: Using asinf may leed to nan (not a number)

the-m-master, 2022-06-09:
Using 'asinf()' may lead to nan. In the function 'FusionQuaternionToEuler' a call is done to 'asinf', if the input of this is larger then 1 or less then -1, the result will be nan. All calculations using this pitch value (as a nan) will fail as well.

A potential solution might be something like:

float asinf_safe(float val)  {
  if (val >= 1.0f) { return (M_PI / 2.0f); }
  if (val <= -1.0f) { return (M_PI / -2.0f); }
  return asinf(val);
}

xioTechnologies, 2022-06-09:
The fix  has been applied as commit [d7788de](https://github.com/xioTechnologies/Fusion/commit/d7788de2114c544d87cba22bd9802dd9244b1d9e).

the-m-master, 2022-06-10:
Thanks! We are going to use it in the next release :)

closed 2022-06-10 by the-m-master