xioTechnologies/Fusion #28: Rotation of global acceleration

viatto, 2022-06-11:
Hi, I am using your library to obtain the acceleration in a global reference frame considered as the initial position of my sensor. So in order to remove the initial rotation of the yaw angle given by the magnetic field, I rotate my acceleration vector of this angle:
`            linacc[0] = linacc[0] * math.cos(angle ) - linacc[1] * math.sin( angle)
              linacc[1] = linacc[0] * math.sin(angle ) + linacc[1] * math.cos( angle) `
Unfortunately, This does not solve the problem can you help to understand the problem? I noticed that if I just translate the sensor in the XY plane, without any rotation, I obtain instability in the yaw measurements.
[image]
 Am I doing something wrong?
This is the code I am using:
`        g = offset.update(g)
        ahrs.update(g, a, MagVals, deltat)
        euler = ahrs.quaternion.to_euler()
        linacc= ahrs.earth_acceleration
        linacc= linacc * 9.81
`
This is the initialization of the algorithm:
`offset = imufusion.Offset(100) #sample rate
ahrs = imufusion.Ahrs()
ahrs.settings = imufusion.Settings(0.5,  # gain
                                   10,  # acceleration rejection
                                   0,  # magnetic rejection
                                   500)  # rejection timeout = 5 seconds`

I tried also with a higher sample rate (180 Hz) and I have better results in the rotation but still unstable yaw.
[image]

xioTechnologies, 2022-06-13:
I suspect that the issue is poor magnetometer calibration.  I suggest you use `ahrs.update_no_magnetometer` instead of `ahrs.update` so that the magnetometer is ignored.  If you are using the algorithm without the magnetometer then you can use the `ahrs.heading = 0` to zero the yaw at any time.

viatto, 2022-06-18:
it is not working the function ahrs.update_no_magnetometer

xioTechnologies, 2022-06-19:
Please provide a description of the issue

viatto, 2022-06-19:
`ahrs.update_no_magnetometer(g1, a1, deltat)` If I try to substitute in the code this function instead of the  classic one  'ahrs.update'. The only result I have in acceleration function `ahrs.earth_acceleration` is a vector of 0 , 0, 1

xioTechnologies, 2022-06-19:
The code and plot below does not demonstrate in the issue you have report.  I believe the problem is with either your data or your use of the library.  I suggest you start from a working example and make small iterative changes towards your intended design.

```python
import imufusion
import matplotlib.pyplot as pyplot
import numpy

data = numpy.genfromtxt("sensor_data.csv", delimiter=",", skip_header=1)

gyroscope = data[:, 1:4]
accelerometer = data[:, 4:7]

ahrs = imufusion.Ahrs()
earth_acceleration = numpy.empty_like(gyroscope)

for index in range(len(gyroscope)):
    ahrs.update_no_magnetometer(gyroscope[index], accelerometer[index], 1 / 100)
    earth_acceleration[index] = ahrs.earth_acceleration

pyplot.plot(earth_acceleration)
pyplot.show()
```
[image]

closed 2022-07-04 by xioTechnologies