xioTechnologies/Fusion #29: Magnetometer Calibration

aaerciyas, 2022-06-13:
We are using "Magneto 1.2" app for calculating soft iron matrix and hard iron offset. The calculated matrix and offset are indicated below.
[image]
To verify calibrated data, we are doing the same process with updated software and expect that soft iron matrix and hard iron offset values are close to zero. However, values are almost the same with uncalibrated matrixes.
[image]

Matrixes are:
`  FusionMatrix soft_iron_matrix_{{{1.160382f, 0.100225f, -0.360193f},`
`                                  {0.100225f, 1.258027f, 0.169535f}, `
`                                  {-0.360193f, 0.169535f, 0.681843f}}};`
`  FusionVector hard_iron_offset_{{55.865986f, -44.026598f, 27.720618f}};`

For the calibrating magnetometer we are calling `FusionCalibrationMagnetic` function:
`magnetometer = FusionCalibrationMagnetic(magnetometer, soft_iron_matrix_, hard_iron_offset_);`

Is there an issue with magnetometer calibration, or am I doing something wrong?

xioTechnologies, 2022-06-13:
Fusion uses the calibration model **m<sub>c</sub> = Sm<sub>u</sub> - h**, as described in the [README](https://github.com/xioTechnologies/Fusion/blob/main/README.md).  Magneto 1.2 uses the calibration model **m<sub>c</sub> = S(m<sub>u</sub> - h)**, as shown in your screenshot.  If you would like to use `FusionCalibrationMagnetic()` with values calculated using Magneto 1.2 then you must recalculate **h** as **Sh**, as shown below.

```c
hard_iron_offset = FusionMatrixMultiplyVector(soft_iron_matrix, hard_iron_offset);
```

aaerciyas, 2022-06-14:
It worked fine! Thank you for your help!

closed 2022-06-14 by aaerciyas