xioTechnologies/Fusion #31: The results of X-aix and Z-axis are affected by Y-axis

JackHenry1992, 2022-06-13:
Thanks for sharing your great job!
In practice, I only rotate the Y axis, but the results of X-axis and Z-axis will also output values, which means that Pitch and Yaw values are affected by the Y-axis.
[image]

The gyroscope value as follows:
[image]

Is there something wrong in my case? I record the data using a samsung mobile phone.

xioTechnologies, 2022-06-13:
Your top graph is mislabelled, you need to swap "Pitch" and "Roll".  The issue you are observing is a well-known problem with Euler angles, commonly referred to as [gimbal lock](https://en.wikipedia.org/wiki/Gimbal_lock).

JackHenry1992, 2022-06-13:
Do you mean that I need to modify the function of `ahrs.quaternion.to_euler()`?

xioTechnologies, 2022-06-13:
No, I am saying that this is just how Euler angles behave.  I suggest you search online to learn more.

JackHenry1992, 2022-06-14:
Thx for your kind reply.

closed 2022-06-14 by JackHenry1992