xioTechnologies/Fusion #33: Gravity acceleration removal

ricber, 2022-06-22:
Hi, thank you for your work! I was wondering if your code removes the gravity acceleration. It seems that it does it, but I'm not sure how. Also, does it need magnetometer data to remove the gravity, or is it sufficient to have accelerometer and gyroscope data? Because I tried the code with my collected data and the gravity was not removed.

xioTechnologies, 2022-06-22:
From the [README](https://github.com/xioTechnologies/Fusion/blob/main/README.md):
"_The algorithm provides three outputs: quaternion, linear acceleration, and Earth acceleration. ... The linear acceleration is the accelerometer measurement with the 1 g of gravity removed. The Earth acceleration is the accelerometer measurement in the Earth coordinate frame with the 1 g of gravity removed._"

A magnetometer is not required for the calculation of linear acceleration or Earth acceleration.

ricber, 2022-06-22:
Does the gravity removal start from the first message or only after an initialization phase? Also, does it require the IMU to be calibrated?

xioTechnologies, 2022-06-22:
You should assume that the algorithm outputs are inaccurate until initialisation is complete.  The accuracy of algorithm outputs will depend on the accuracy of the sensor measurements you provide.  [Garbage in, garbage out](https://en.wikipedia.org/wiki/Garbage_in,_garbage_out).

ricber, 2022-06-27:
Hi! I tried your filter but the gravity removal it doesn't seem to be happening. Do you know which are possible causes of this behavior?

ricber, 2022-06-28:
Hi, I am trying to do dead reckoning by using an adaptation of your `gait_tracking.py` script. I removed the detection of the moving periods and used the `linear_acceleration` instead of the `earth_acceleration`. For my use case, it is important to know the transformation between successive positions accurately and not have an accurate long-run positioning. So I can ignore the drift. I run the script with my IMU data, but the plot of the linear_acceleration reveals that gravity has not been removed on the Z component. 

My script:
`
from matplotlib import animation
import imufusion
import matplotlib.pyplot as pyplot
import numpy

# Import sensor data ("short_walk.csv" or "long_walk.csv")
data = numpy.genfromtxt("my_walk.csv", delimiter=",", skip_header=1)

sample_rate = 110  # Hz

timestamp = data[:, 0]
gyroscope = data[:, 1:4]
accelerometer = data[:, 4:7]

# Plot sensor data
figure, axes = pyplot.subplots(nrows=6, sharex=True, gridspec_kw={"height_ratios": [6, 6, 6, 2, 1, 1]})

figure.suptitle("Sensors data, Euler angles, and AHRS internal states")

axes[0].plot(timestamp, gyroscope[:, 0], "tab:red", label="Gyroscope X")
axes[0].plot(timestamp, gyroscope[:, 1], "tab:green", label="Gyroscope Y")
axes[0].plot(timestamp, gyroscope[:, 2], "tab:blue", label="Gyroscope Z")
axes[0].set_ylabel("Degrees/s")
axes[0].grid()
axes[0].legend()

axes[1].plot(timestamp, accelerometer[:, 0], "tab:red", label="Accelerometer X")
axes[1].plot(timestamp, accelerometer[:, 1], "tab:green", label="Accelerometer Y")
axes[1].plot(timestamp, accelerometer[:, 2], "tab:blue", label="Accelerometer Z")
axes[1].set_ylabel("g")
axes[1].grid()
axes[1].legend()

# Intantiate AHRS algorithms
offset = imufusion.Offset(sample_rate)
ahrs = imufusion.Ahrs()

ahrs.settings = imufusion.Settings(0.5,  # gain
                                   10,  # acceleration rejection
                                   0,  # magnetic rejection
                                   5 * sample_rate)  # rejection timeout = 5 seconds

# Process sensor data
delta_time = numpy.diff(timestamp, prepend=timestamp[0])

euler = numpy.empty((len(timestamp), 3))
internal_states = numpy.empty((len(timestamp), 3))
acceleration = numpy.empty((len(timestamp), 3))

for index in range(len(timestamp)):
    gyroscope[index] = offset.update(gyroscope[index])

    ahrs.update_no_magnetometer(gyroscope[index], accelerometer[index], delta_time[index])

    euler[index] = ahrs.quaternion.to_euler()

    ahrs_internal_states = ahrs.internal_states
    internal_states[index] = numpy.array([ahrs_internal_states.acceleration_error,
                                          ahrs_internal_states.accelerometer_ignored,
                                          ahrs_internal_states.acceleration_rejection_timer])

    acceleration[index] = ahrs.linear_acceleration

# Plot Euler angles
axes[2].plot(timestamp, euler[:, 0], "tab:red", label="Roll")
axes[2].plot(timestamp, euler[:, 1], "tab:green", label="Pitch")
[... 93 lines cut ...]
        if (min(x) != max(x)) and (min(y) != max(y)) and (min(z) != max(z)):
            axes.set_xlim3d(min(x), max(x))
            axes.set_ylim3d(min(y), max(y))
            axes.set_zlim3d(min(z), max(z))

            axes.set_box_aspect((numpy.ptp(x), numpy.ptp(y), numpy.ptp(z)))

        return scatter

    anim = animation.FuncAnimation(figure, update,
                                   frames=int(len(timestamp) / samples_per_frame),
                                   interval=1000 / fps,
                                   repeat=False)

    anim.save("animation.gif", writer=animation.PillowWriter(fps))

pyplot.show()
`

My data: 
[my_walk.csv](https://github.com/xioTechnologies/Fusion/files/9001353/my_walk.csv)

Plots:
[image]
[image]

xioTechnologies, 2022-06-28:
The problem is that you are providing accelerometer measurements in m/s/s, not g.  The algorithm expects accelerometer measurements in g as described by the documentation, `* @param accelerometer Accelerometer measurement in g.` on [line 96](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L96).

In addition to this documentation:
- You modified the [example code](https://github.com/xioTechnologies/Gait-Tracking/blob/main/gait_tracking.py#L64) to replace `acceleration[index] = 9.81 * ahrs.earth_acceleration  # convert g to m/s/s` with `acceleration[index] = ahrs.linear_acceleration`, deleting a comment that clarifies the units of measurement.
- You opened a [separate issue](https://github.com/xioTechnologies/Gait-Tracking/issues/7) to ask about this line of code where I clarified that the units are g and not m/s/s.
- Your plot above shows the accelerometer is measuring 10 g while stationary - a obvious hint that you are using m/s/s when g is expected.

You also need to modify your code above to use `earth_acceleration` as per the example code, not `linear_acceleration`.

ricber, 2022-06-28:
@xioTechnologies I see, so I have to divide my accelerometer data by 9.81. Thank you! I removed the comment from the code because I thought the `linear_acceleration` was in m/s/s. As far as I have understood, the earth acceleration is relative to the earth reference frame, while the liner acceleration is relative to what?

ricber, 2022-06-29:
Thank you for being so helpful! I think it is a good idea to update the documentation. I also suggest updating the [README](https://github.com/xioTechnologies/Fusion#readme) to explain better that by **g**, you mean a unit of measurement. Since **g** is not part of the [SI units](https://en.wikipedia.org/wiki/International_System_of_Units), it is not immediate to get its meaning for a non-expert of the subject.

closed 2022-06-29 by ricber

Sinou12, 2022-06-29:
Hello, 
thank you for your feedback. 
I know that this code is not made for this kind of movement. 
But from your experience, I would like to know your opinion.

I have done a 20 cm displacement along the Y axis only, and I have guaranteed that there will be no rotation along the 3 axes.
The gyroscope displays 0 deg/s in all 3 axes. 

To extract the gravity from the acceleration to calculate the displacement along the Y axis.  Do I just need a filter to extract it? 
Or according to the Y axis, I won't need to do it, because the accelerometer displays 1 g according to Z only?
Since the rotation speed is equal to 0 deg/s, so the rotation matrix is zero, so I don't need to calculate the acceleration in the terrestrial frame (I mean it's the same in both frames)? 

Thank you.

xioTechnologies, 2022-06-29:
@ricber I do not recognise your issue with g.  The use of g as a unit of measurement is common practise.  It is even used in common parlance.  For example, when talking about rolla coasters, motor sports, aeroplanes, etc.  Most importantly, g is the conventional unit of measurement for accelerometers.  If you check the datasheet of the device you are using, I expect you will see g being used, and m/s/s not mentioned at all.

SI units are irrelevant to convention.  Kelvin is SI but °C is conventional, radians is SI but degrees is conventional, V/A is SI but Ω is conventional, and so on.

ricber, 2022-06-29:
@xioTechnologies thank you for the feedback! I understand your point but IMUs always provide the measurements in m/s/s. Also in ROS, m/s/s is the standard unit of measure for the [IMU message](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/Imu.html) and they explicitly say “ not in g's”.

Taeyoung96, 2022-10-10:
@ricber   
Hi like you, I am going to use ROS IMU message and use dead reconning with the Fusion algorithm.  
To change the acceleration unit, we did the following process, but it didn't work much.  
```cpp
const sensor_msgs::Imu thisImu = *imu_msg;
float delta_time = (thisImu.header.stamp - estimate_timestamp_).toSec();    // unit : s

auto acc = thisImu.linear_acceleration;  // unit : m/s^2
FusionVector accelerometer = {float(acc.x / 9.805f) , float(acc.y / 9.805f), float(acc.z / 9.805f)}; 

auto gyr = thisImu.angular_velocity;  // unit : rad/s
FusionVector gyroscope = {rad2deg(gyr.x), rad2deg(gyr.y), rad2deg(gyr.z)}; 

FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, delta_time);

```
How did you solve the problem?