xioTechnologies/Fusion #35: Suggestion on calculating the calibration parameters

ryanhz, 2022-06-30:
As I read on README:
`This library does not provide a solution for calculating the calibration parameters.`

Do you have any suggestion on how to calculate calibration parameters, especially misalignment and offset?
The IMU is installed in an unknown orientation.

xioTechnologies, 2022-07-01:
We do not have any tutorials for simple solutions for calibration.  I suggest you search on-line.  I'm sorry I cannot be of more help.

closed 2022-07-01 by xioTechnologies

ryanhz, 2022-07-01:
Thank you for your prompt reply.  I have figured it out.  I was not clear the meaning of the elements of the misalignment matrix previously.

LazaroFilm, 2023-01-17:
I am also unclear about the misalignment argument. Can someone shine a light on what it is and how to generate it?