xioTechnologies/Fusion #36: Fusion calibration matrices

ajaybhargav, 2022-07-04:
I have questions regarding the matrices used in following equation:

i<sub>c</sub> = Ms(i<sub>u</sub> - b)

I understand i<sub>u</sub> and b.

How to calculate M for accelero and gyro?
What are is input format for M?

is s (sensitivity matrix) sensitivity of sensor or calibration model?

xioTechnologies, 2022-07-04:
As stated the in the [README](https://github.com/xioTechnologies/Fusion), "_This library does not provide a solution for calculating the calibration parameters._".  The expanded inertial calibration model equation is shown below, perhaps this clarifies things.

[image]

ajaybhargav, 2022-07-04:
I am really sorry. Yes i read that. I read some papers related to the equation and I got understanding a bit. I am just confused on calculating M, rest all I figured out.

Also in the image you posted, s is a diagonal matrix of 3x3, whereas in source (https://github.com/xioTechnologies/Fusion/blob/main/Examples/Advanced/main.c) it looks like a 3x1 matrix. I am sorry if I am being too naive with my queries.

xioTechnologies, 2022-07-04:
The diagonal matrix contains only 3 non-zero elements and so can be expressed as a vector of 3 elements.

closed 2022-08-02 by xioTechnologies