xioTechnologies/Fusion #37: On Gyroscope ma

David-OConnor, 2022-07-05:
Related to #16.

The AHRS algorithm takes gyroscope measurements as 3-point vectors:
```c
void FusionAhrsUpdate(FusionAhrs *const ahrs, const FusionVector gyroscope, const FusionVector accelerometer,
```

These vectors use x, y, and z, in that order internally:
From FusionMath.h:
```c
typedef union {
    float array[3];

    struct {
        float x;
        float y;
        float z;
    } axis;
} FusionVector;
```

The official examples imply (And my testing implies as well) the order in these vecs is **roll**, **pitch**, **yaw**.
From official examples:
```c
const FusionEuler euler = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
printf("Roll %0.1f, Pitch %0.1f, Yaw %0.1f\n", euler.angle.roll, euler.angle.pitch, euler.angle.yaw);
```

**Point 1:** This implies Gyro X -> roll, gyro Y -> pitch, gyro Z -> yaw

Image from a common IMU, that matches intuition:
[image]

Point 2: **This would imply pitch is movement *around the x axis* and reported in IMU gyro X registers, and roll is movement *around the y axis* and reported in IMU gyro Y registers. **

How do we reconcile these opposite pitch and roll mappings? Thank you.

xioTechnologies, 2022-07-05:
You need to stop conflating Euler angles with gyroscope axis.  The gyroscope axes are X, Y, and Z.  The terms roll, pitch, and yaw cannot not be used to describe the gyroscope axes.  You said in the [related issue](https://github.com/xioTechnologies/Fusion/issues/16) that "_this is convention I've been using_".  You need to abandon this convention.  It is incorrect and causing you confusion.

David-OConnor, 2022-07-05:
Thanks for getting back to me! I think we're on the same page: gyroscope input being a struct with *X*, *Y*, and *Z* labels. In the code base, there's no indication of which Euler angle matches to which struct field. This is important

Stated simply: *What order do you need to 

I think this could be solved with one or more of these approaches:

- #1: Adding to the `FusionAhrsUpdate` doc comment the order to pass gyroscope measurements. (https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L92)

- #2: Modifying one of the examples to be more explicit. For example, replace this:
```c
const FusionVector gyroscope = {0.0f, 0.0f, 0.0f}; // replace this with actual gyroscope data in degrees/s
```

with this:
```c
// Replace `roll`, `pitch`, and `yaw` with actual gyroscope data in degrees/s.
const FusionVector gyroscope = {roll, pitch, yaw};
```

#3: Accept a [FusionEuler](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionMath.h#L444) struct instead of `FusionVector` for gyroscope inputs to `FusionAhrsUpdate`.

xioTechnologies, 2022-07-05:
You believe there is a problem because you have a fundamental misunderstanding of Euler angles.  The solution is for you to correct this misunderstanding, not modify the library.

To repeat, "_Pitch, roll, and yaw are Euler angles. These terms are not applicable to gyroscope measurements of angular velocity._", and "_The terms roll, pitch, and yaw cannot not be used to describe the gyroscope axes._".

David-OConnor, 2022-07-05:
I understand that.

The order matters in terms of fusing the accel and gyro measurements. The API and docs are ambiguous about which order you pass gyro measurements into the *vector struct thats fields are named X, Y, and Z*

Thought experiment: Let's say I store my gyro reading euler angles as variables `pitch`, `roll`, and `yaw`:

I run this to perform an attitude update:
```c
        const FusionVector gyroscope = {yaw, pitch, roll};
      
        FusionAhrsUpdateNoMagnetometer(&ahrs, gyroscope, accelerometer, SAMPLE_PERIOD);
```

Should I expect accurate results?

What if I substitute this line:
```c
const FusionVector gyroscope = {pitch, pitch, pitch};
```

This
```c
const FusionVector gyroscope = {pitch, roll yaw};
```

Re your last line, in an attempt to clarify:
*Pitch, roll, and yaw are Euler angles... The API uses the `Vec {x, y, z}` type, not the `EulerAngle` type for updating the algorithm, and gives no indication of in which order to pass the measurements. These must be consistent, or they will fuse incorrectly with the other terms.*

closed 2022-07-05 by xioTechnologies

WelinLee, 2022-10-11:
Hi, the discussion makes more confusion. I would like to explain more simplified.
The measurement from gyroscope is regarding to the sensor frame, whilst Euler angles indicate the rotation from current sensor frame regarding to the reference frame. That's why x-io always emphasizes "_**Pitch, roll, and yaw are terms not applicable to gyroscope measurements of angular velocity**_".
In the Fusion algorithms, the reference frame is Earth frame by **NWU** convention.
If sensor frame setting by manufacture is not matched with **NWU** convention, see [FusionAxesSwap](https://github.com/xioTechnologies/Fusion/blob/bbbdd630e031d183db85e698b2dff573f683faef/Fusion/FusionAxes.h#L60) functions in detail.

David-OConnor, 2023-05-22:
Fixed by https://github.com/xioTechnologies/Fusion/issues/9