xioTechnologies/Fusion #42: Weird wrong orientation obtained

aidengk-iris, 2022-08-08:
Dear xioTechnologies and all,

I am using the Fusion library to get an quaternion description of the global orientation of my 9DOF IMU (accelerometer, magnetometer and gyroscope. All axes aligned). 

Since I am quite new to orientations, I followed the advanced example from the library and used Adafruit's 3D orientation visualizer to get a sense of the orientation. 

The accelerometer and the magnetometer have been calibrated. Their calibration data is entered in the code, as instructed in the example. 

However, the orientation I get seemed wrong and unstable. Please see the video included for a demo of the issue. [https://youtu.be/hq4xZufC_eM](url)

I have also included the code, hoping it helps identifying the issue. 

The sudden change in orientation can be the result of magnetic rejection kicking in. I wish I could come up with more possible causes but unfortunately I am out of ideas. 

I am grateful to any insight on the problem. Many thanks.

Hardware info:

- Microcontroller used: SONY SPRESENSE
- IDE: Arduino
- Gyroscope : FXAS21002C
- Accelerometer and Magnetometer : FXOS8700

My code: 
```
#include "Fusion.h"
#include <Adafruit_FXOS8700.h>
#include <Adafruit_FXAS21002C.h>
#include <Adafruit_Sensor.h>
#include <Wire.h>

#define BAUD_RATE             115200
#define G_VALUE               9.81f
#define DEGREES_PER_RADIAN    57.29578
#define PI                    3.1415926
#define SAMPLE_RATE           100   // gyroscope set to 100Hz output rate

/* Assign a unique ID to this sensor at the same time */
Adafruit_FXAS21002C gyro = Adafruit_FXAS21002C(0x0021002C);
Adafruit_FXOS8700 accelmag = Adafruit_FXOS8700(0x8700A, 0x8700B);

// Define calibration
const FusionMatrix gyroscopeMisalignment = {  1.0f, 0.0f, 0.0f, 
                                              0.0f, 1.0f, 0.0f, 
                                              0.0f, 0.0f, 1.0f  };

const FusionVector gyroscopeSensitivity = {1.0f, 1.0f, 1.0f};

const FusionVector gyroscopeOffset = {0.0f, 0.0f, 0.0f};        // the zero rate is super low, so I assume it's just zeros

const FusionMatrix accelerometerMisalignment = {  1.0f, 0.0f, 0.0f, 
                                                  0.0f, 1.0f, 0.0f, 
                                                  0.0f, 0.0f, 1.0f  };

const FusionVector accelerometerSensitivity = {1.0f, 1.0f, 1.0f};

const FusionVector accelerometerOffset = {0.02f, 0.02f, 1.02f};

const FusionMatrix softIronMatrix = { 1.007f, 0.015f, 0.007f, 
                                      0.015f, 0.931f, 0.01f, 
                                      0.007f, 0.01f, 1.067f      };

const FusionVector hardIronOffset = {-21.7f, -44.64f, -63.13f};

FusionOffset offset;
FusionAhrs ahrs;
// Set AHRS algorithm settings
const FusionAhrsSettings settings = {
        .gain = 0.5f,
        .accelerationRejection = 10.0f,
        .magneticRejection = 20.0f,//60.0f,
        .rejectionTimeout = 5 * SAMPLE_RATE , /* 5 seconds */
};

/*  declare sensor events  */
sensors_event_t   acc_event;
sensors_event_t   gyro_event;
sensors_event_t   mag_event;

float time_now = 0.0f;
float time_pre = 0.0f;

void setup() {

  Wire.begin();
  Serial.begin( BAUD_RATE );
  Wire.setClock(400000); // 400KHz

    /* Initialise the sensors */
  while(!accelmag.begin())
  {
    Serial.println(F("Connection with Accelerometer and Magnetometer NOT established"));
    delay(3000);
  }
    
  Serial.println("Accelerometer and Magnetometer initialized");

  while(!gyro.begin())
  {
    Serial.print("Connection with Gyro NOT established ");
    delay(3000);
  }

  Serial.println("Gyroscope initialized");

  // set range
  gyro.setRange( GYRO_RANGE_500DPS );
  accelmag.setAccelRange(ACCEL_RANGE_2G);

  Serial.print("Acclerometer Range: ");  Serial.print( accelmag.getAccelRange()); Serial.println(" ");
  Serial.print("Acc Mag Hybrid output rate: "); Serial.println( accelmag.getOutputDataRate() );
  Serial.print("Gyro Range: ");  Serial.print( gyro.getRange() ); Serial.println(" ");

  // Initialise algorithms
  FusionOffsetInitialise(&offset, SAMPLE_RATE);
  FusionAhrsInitialise(&ahrs);
  FusionAhrsSetSettings(&ahrs, &settings);
}

void loop(void) {

  /* update sensor events */
  accelmag.getEvent(&acc_event, &mag_event);
  gyro.getEvent(&gyro_event);

  // Acquire latest sensor data
  FusionVector gyroscope    = {   gyro_event.gyro.x  , 
                                  gyro_event.gyro.y  , 
                                  gyro_event.gyro.z        }; // actual gyroscope data in degrees/s
  FusionVector accelerometer = {  acc_event.acceleration.x, 
                                  acc_event.acceleration.y, 
                                  acc_event.acceleration.z  }; // actual accelerometer data in g
  FusionVector magnetometer = {   mag_event.magnetic.x, 
                                  mag_event.magnetic.y, 
                                  mag_event.magnetic.z      }; // actual magnetometer data in uTesla
  
  time_now = micros(); // replace this with actual gyroscope timestamp

  // Apply calibration
  gyroscope     = FusionCalibrationInertial(gyroscope, gyroscopeMisalignment, gyroscopeSensitivity, gyroscopeOffset);
  accelerometer = FusionCalibrationInertial(accelerometer,  accelerometerMisalignment, accelerometerSensitivity, accelerometerOffset);
  magnetometer  = FusionCalibrationMagnetic(magnetometer,   softIronMatrix,   hardIronOffset);

  // Update gyroscope offset correction algorithm
  gyroscope = FusionOffsetUpdate(&offset, gyroscope);

  float deltaTime = (time_now - time_pre) / 1000000.0f;  // time difference in seconds
  time_pre = time_now;

  // Update gyroscope AHRS algorithm
  FusionAhrsUpdate(&ahrs, gyroscope, accelerometer, magnetometer, deltaTime);
  // Serial.println(deltaTime);

  // get quaternion rotation
  FusionQuaternion q = FusionAhrsGetQuaternion(&ahrs);

  Serial.print("Quaternion: ");
  Serial.print(q.element.w);
  Serial.print(", ");
  Serial.print(q.element.x);
  Serial.print(", ");
  Serial.print(q.element.y);
  Serial.print(", ");
  Serial.println(q.element.z);

  delay(10);
}
```

xioTechnologies, 2022-08-08:
I suggest the first step is to solving this is to use the [x-IMU3 GUI](https://x-io.co.uk/x-imu3/) to display the 3D orientation and sensor data side-by-side, in real-time.  You will need to modify your firmware to send the following packets by serial.  See page 23 of the [x-IMU3 user manual](https://x-io.co.uk/downloads/x-IMU3-User-Manual-v0.11.pdf) for more information about these packets.

- `I,1000000,0.0000,0.0000,0.0000,0.0000,0.0000,1.0000\r\n` (gyroscope and accelerometer)
- `M,1000000,1.0000,0.0000,0.0000\r\n` (magnetometer)
- `Q,1000000,1.0000,0.0000,0.0000,0.0000\r\n` (quaternion)

If your firmware is sending these packets then you can manually open a serial connection to view data in real-time as shown in the screenshot below.

[image]

aidengk-iris, 2022-08-09:
@xioTechnologies 
Thank you very much for the suggestion. The GUI made locating the issue a lot easier! 

The issue turns out to be a wrong value in the accelerometer's offset. I put 1.03g instead of 0.03g in the z axis offset. 

Of course, as the offset is subtracted from the acceleration readings, the gravity reference is very shaky. 

I do have another question that is not closely related to this issue, so I will discuss it in a new post.

closed 2022-08-09 by aidengk-iris