xioTechnologies/Fusion #43: Slowly getting to the correct orientation

aidengk-iris, 2022-08-09:
Dear xioTechnologies,

Thanks to the X-IMU3 GUI, I was able to visualize the orientation of my 9DOF IMU (Acclerometer, magnetometer and gyro).

I am using the default parameters for the algorithm's initialization:
- gain = 0.5
- accelerationRejection = 10 degrees
- magneticRejection = 20 degrees
- rejectionTimeout = 5 * 200 samples , /* stands for 5 seconds */. The gyro sampling rate is set to 200 Hz.

However, as I rotated the IMU, the measured orientation slowly approached the correct orientation but not exactly right. Usually the measured orientation became correct after a reinitialization triggered by the timeout. 

I also tried disabling the magnetic rejection but had similar observations. I demonstrated this issue in a short video. 
[https://youtu.be/yuEquhkZYPk](url) 

My guess is that the rejection parameters need to be tuned, or maybe disabled? The application I'm working on involves constant rotation of the imu, not much accleration though. I wonder what I should do to obtain accurate orientation measurements, for example like in the video [https://youtu.be/BXsGWoOMtmU](url)

Many thanks for your insight.

Finally my hardware info, thought I don't think it matters:

Microcontroller used: SONY SPRESENSE. It has 6 ARM® Cortex®-M4F cores running at 156 MHz
IDE: Arduino
Gyroscope : FXAS21002C
Accelerometer and Magnetometer : FXOS8700

xioTechnologies, 2022-08-10:
You have shown that the 3D model initially only rotates half the expected angle.  This will be due to an incorrect gyroscope sensitivity or `deltaTime` value.  Perhaps your `deltaTime` value is half what it should be.

I suggest that you start with a simple implementation and gradually increase complexity.  I suggest that you implement each of the following systems and only progress to the next once the current system is working without issue.

1. Gyroscope and accelerometer only.  Acceleration rejection disabled.
2. Gyroscope and accelerometer only. Acceleration rejection enabled.
3. Gyroscope, accelerometer, and magnetometer.  Acceleration rejection enabled.  Magnetic rejection disabled.
4. Gyroscope, accelerometer, and magnetometer.  Acceleration rejection enabled.  Magnetic rejection enabled.

aidengk-iris, 2022-08-11:
@xioTechnologies 
Many thanks to your suggestions! I am testing indivisual systems as you suggested. 
Already in the gyro and accelerometer only case (no acc rejection), I can see you are right that there is something wrong with the gyro's sensitivity. 
Setting the gyro's sensitivity vector to 2s resulted correct and very stable orientation estimation. But this is a test fix just for debugging, I will double check the sensor's driver library for errors. 

Thanks again. I will close the issue.

closed 2022-08-11 by aidengk-iris