xioTechnologies/Fusion #47: magnetometer unit

rohithr2597, 2022-09-23:
Hi. I am trying to visualize the IMU data and I referred to this issue (https://github.com/xioTechnologies/Fusion/issues/42) to send data serially to the X-IMU GUI. IMU Visualization is changing its orientation faster and it's not stable. It's moving randomly even though the IMU is stable. So, From the manual, the unit of accelerometer data is G and gyroscope data is DPS. Unit of magnetometer is given as a.u. so I have magnetometer output in gauss. So, Do I need to convert it into microtesla before transferring it serially?

xioTechnologies, 2022-09-23:
The magnetometer units do not matter provided that the magnetometer is calibrated.  I suggest you start with a simple implementation and gradually increase complexity, as described in [this comment](https://github.com/xioTechnologies/Fusion/issues/43#issuecomment-1210620225).

aidengk-iris, 2022-09-28:
@rohithr2597  Following the suggestion from xioTechnologies, I would also doubleck check if the IMU has been properly calibrated for all the sensors on it. And also double check the detection ranges of the sensors are properly set.

rohithr2597, 2022-09-28:
Thanks for your help @aidengk-iris and @xioTechnologies. Actually, the problem was with the register initialization of the IMU. Now I am able to visualize in the GUI with a slight drift. Initially, I got confused with the magnetometer unit. But now it's fine.

closed 2022-09-28 by rohithr2597