xioTechnologies/Fusion #48: Somewhat confused by the Euler angle conversion

kamnxt, 2022-10-06:
Hi!

I'm having some issues with understanding the relation between the imu input axes and the euler angles from the resulting quaternion.
Here's an example: I have equal acceleration in the Y and the Z axis. I would guess that in this case I should be getting 45 degrees of pitch, but instead I am getting 45 degrees of roll? I was expecting to have Z upwards, X to the side and Y in the front/rear axis, but it seems that isn't the case here?

```
#include "Fusion/Fusion.h"
#include <stdio.h>

int main() {
	FusionAhrs ahrs;
	FusionAhrsInitialise(&ahrs);
	for(int i = 0; i < 100; i++) {
		FusionVector acc, gyr;
		acc.axis.x = 0;
		acc.axis.y = 1;
		acc.axis.z = 1;
		gyr.axis.x = 0;
		gyr.axis.y = 0;
		gyr.axis.z = 0;
		FusionAhrsUpdateNoMagnetometer(&ahrs, gyr, acc, 0.01);
	    FusionEuler eul = FusionQuaternionToEuler(FusionAhrsGetQuaternion(&ahrs));
	    printf("p,r,y = %.3f, %.3f, %.3f\n", eul.angle.pitch, eul.angle.roll, eul.angle.yaw);
            //prints p,r,y = 0.000, 45.012, 0.000
	}
}
```

xioTechnologies, 2022-10-06:
The Euler angles [documentation](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionMath.h#L68) states, "_Roll, pitch, and yaw correspond to rotations around X, Y, and Z respectively._".  The behaviour you have described is consistent with the documentation.

xioTechnologies, 2022-10-06:
The [README](https://github.com/xioTechnologies/Fusion#readme) already states, "_The quaternion describes the orientation of the sensor relative to the Earth using the North-West-Up (NWU) convention._".  That is, X = North, Y = West, Z = Up.