xioTechnologies/Fusion #51: Question: about the output quaternion regarding to "North-West-Up (NWU)"

mingrenzhu, 2022-11-11:
hello, can you explain why the output quaternion is based on the North-West-Up (NWU) ? I don't have no idea about this discription, " The quaternion describes the orientation of the sensor relative to the Earth using the North-West-Up (NWU) convention" . And I want wo know the principle, can you geve me some tips about this ? thanks a lot

xioTechnologies, 2022-11-11:
The North-West-Up (NWU) describes the axis directions relative to the Earth.  X=North, Y=West, Z=Up.

mingrenzhu, 2022-11-11:
thanks for your reply, I means why to set the North-West-Up coordinate. In genera, we can use the one piece of acc and mag data to calculate transformation matrix based on ENU or NWU between the earth coordinate and imu coordinate, and set this transformation as the intial pose q0. In fusion algorithm, we set the initial quaternion pose is identity matrix, and the right pose will be right gradually with the feed of acc and mag data, that is why the "ahrs->rampedGain" coefficients is lager, because we use the acc and mag data to converge into right pose. and during this convergence process, we make sure the NWU coordinate, not ENU coordinate, is the viewpoint right ?
      following codes are related with the decision with the NWU coordinate, is it right ?
```
   const FusionVector halfGravity = {
            .axis.x = Q.x * Q.z - Q.w * Q.y,
            .axis.y = Q.y * Q.z + Q.w * Q.x,
            .axis.z = Q.w * Q.w - 0.5f + Q.z * Q.z,
    };
   ahrs->halfAccelerometerFeedback = FusionVectorCrossProduct(FusionVectorNormalise(accelerometer), halfGravity);
   const FusionVector halfWest = {
                .axis.x = Q.x * Q.y + Q.w * Q.z,
                .axis.y = Q.w * Q.w - 0.5f + Q.y * Q.y,
                .axis.z = Q.y * Q.z - Q.w * Q.x
        };
    ahrs->halfMagnetometerFeedback = FusionVectorCrossProduct(FusionVectorNormalise(FusionVectorCrossProduct(halfGravity, magnetometer)), halfWest);
```

xioTechnologies, 2022-11-11:
I am sorry but I do not understand your description and question.  NWU and ENU are conventions.  Fusion uses NWU.

mingrenzhu, 2022-11-11:
Sorry for confusing you, i change my question to another. if I want to change the output quaternion pose to be based on ENU ? what should i do ?

mingrenzhu, 2022-11-11:
> I am sorry but I do not understand your description and question. NWU and ENU are conventions. Fusion uses NWU.

Sorry for giving this uncomfortable understanding for you. I want to give further explanations for you. In your thesis, you mentioned that your output is based the North-West coordinate. My question is that which step in your algorithm is based on the North-East coordinate. 

Thanks for your patience and hope your reply.  Cheers!

xioTechnologies, 2022-11-11:
A change from NWU to ENU would only affect [line 164](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L164).  If you are only using a gyroscope and accelerometer then no change is required.  In this case, NWU and ENU are identical because the algorithm would have no reference of heading.

mingrenzhu, 2022-11-11:
I see, and if i want to change the NWU to ENU, how can i revise the [line 164](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L164) to change into ENU ? can you give me an example or some related materials ? 

Thanks for your patience and hope your reply. Cheers!

xioTechnologies, 2022-11-11:
My previous comment was incorrect, several other changes would be required.  The changes are too complex to guide you through here.  I will consider this a feature request for Fusion to support NWU, NED, and ENU.  I cannot suggest a completion date at this time.  However, you can access the WIP once the feature branch is available.

mingrenzhu, 2022-11-11:
Cool, thanks a lot.

xioTechnologies, 2022-11-14:
ENU and NED support has been added in [axes-convention](https://github.com/xioTechnologies/Fusion/tree/axes-convention).

mingrenzhu, 2022-11-14:
Wonderful !! very thankful !!

mingrenzhu, 2022-11-14:
One more question, i want to know the theory behind the Ned or END calculation code, can you give me some documents or advice about that ?  thanks.

mingrenzhu, 2022-11-14:
Hello, I check the [axes-convention](https://github.com/xioTechnologies/Fusion/tree/axes-convention) branch code, and discover that the code realization about function named by "HalfGravity(const FusionAhrs *const ahrs)" is not complete. the "case FusionConventionNwu"  is not realized.

`static FusionVector HalfGravity(const FusionAhrs *const ahrs) {
#define Q ahrs->quaternion.element
    switch (ahrs->settings.convention) {
        case FusionConventionNwu:
        case FusionConventionEnu:
            return (FusionVector) {
                    .axis.x = Q.x * Q.z - Q.w * Q.y,
                    .axis.y = Q.y * Q.z + Q.w * Q.x,
                    .axis.z = Q.w * Q.w - 0.5f + Q.z * Q.z,
            }; // third column of transposed rotation matrix scaled by 0.5
        case FusionConventionNed:
            return (FusionVector) {
                    .axis.x = -1.0f * (Q.x * Q.z - Q.w * Q.y),
                    .axis.y = -1.0f * (Q.y * Q.z + Q.w * Q.x),
                    .axis.z = -1.0f * (Q.w * Q.w - 0.5f + Q.z * Q.z),
            }; // third column of transposed rotation matrix scaled by -0.5
        default:
            return FUSION_VECTOR_ZERO; // avoid compiler warning
    }
#undef Q
}`

xioTechnologies, 2022-11-14:
North-West-Up (NWU), East-North-Up (ENU), and North-East-Down (NED) describe the alignment of the Earth XYZ axes.  I do not have any documentation for how calculations differ for each.  It should be clear from the code and comments, assuming the reader is familiar with the underlying mathematics.  For example, the acceleration of gravity is in the up direction.  For NWU and ENU, up is aligned with +Z and so gravity must be equal to the third column of a rotation matrix.  For NED, up is aligned with -Z and so gravity must be equal to the negated third column of a rotation matrix. 

The code is complete.  The value of `halfGravity` is the same for `FusionConventionNwu` and `FusionConventionEnu` so the switch statement implements a [fall through](https://c-for-dummies.com/blog/?p=3607).

mingrenzhu, 2022-11-15:
Thanks for your reply. I check the theory from chapter 7 of [Madgwick's PhD thesis](https://ethos.bl.uk/OrderDetails.do?uin=uk.bl.ethos.681552) combined with your explanation, and i understand the "HalfGravity" vector calculation based on different axe conversions. thanks again.

closed 2023-01-09 by xioTechnologies