xioTechnologies/Fusion #52: Question: about the imu data(acc, gyro, mag)  which is not desynchronized

mingrenzhu, 2022-11-11:
hello, I have a  question about " can the fution algorithm process the desynchronized imu data ? ", thanks a lot.

xioTechnologies, 2022-11-11:
If your sensor measurements are not synchronised then you would call `FusionAhrsUpdate` for each new gyroscope measurement, with with most recent measurements from the other sensors.

mingrenzhu, 2022-11-11:
well, i wonder whether i can do this, i call **FusionAhrsUpdate** for the new gyroscope, and if gyroscope time is large with acc and mag time, than can i set the acc and mag as zeros vector ? 
thank you and hope your reply, cheers!

xioTechnologies, 2022-11-11:
You should use the latest accelerometer and magnetometer measurements, even if that means reusing measurements.  You should not pass in zero vectors.

mingrenzhu, 2022-11-12:
Got it, thanks a lot.

closed 2023-01-09 by xioTechnologies