xioTechnologies/Fusion #55: Euler angles problem

machurapiotrek, 2022-11-23:
I'm using MPU6500 to receive raw data form gyro and accelerometer. Sampling frequency is set at 500Hz.

I'm using your python gait-tracking example to data visualisation. With your short_walk.csv file, algorithm works well. 

I can't obtain any good results with my data.
I tilted MPU about 45 degrees and backed to starting position.

As you can see pitch value didn't return to starting position but rose.
I can't figure out what's wrong.

I attach my sample file log.csv. 

[log.csv](https://github.com/xioTechnologies/Fusion/files/10074565/log.csv)
[image]
[image]

xioTechnologies, 2022-11-24:
In your plot, the angular velocities measured by the gyroscope are in the opposite direction to the changes in inclination indicated by the accelerometer.  It is not clear if you have negated gyroscope measurements, or if the gyroscope axes are misaligned to the accelerometer.

The following two plots below were generated using your data.  The first plot uses negated gyroscope measurements.  The second plot adjusts the gyroscope alignment.  Note that the algorithms determines a lower acceleration error in the second plot.  This suggests that the issue is misalignment and not negation.

[image]

[image]

The gyroscope alignment was adjusted in the second plot using the following line of code, inserted immediately before [line 53](https://github.com/xioTechnologies/Gait-Tracking/blob/main/gait_tracking.py#L53) in [gait_tracking.py](https://github.com/xioTechnologies/Gait-Tracking/blob/main/gait_tracking.py).  Similar results are achieved using `NXNYPZ`, `PZNYPX`, and `NZNYNX`.  You will have to review your design to determine what the actual misalignment it.

`gyroscope[index] = imufusion.axes_swap(gyroscope[index], imufusion.ALIGNMENT_PXNYNZ)`

`.

machurapiotrek, 2022-11-24:
Thank you for the fast response! I can see clearly now that I need to study the basics of this subject.

closed 2022-11-24 by machurapiotrek