xioTechnologies/Fusion #57: pose derivation in FusionAhrsUpdate algorithm

jacklee0575, 2023-01-09:
Recently, I'm learning your fusion algorithm. Thanks for your great work!
I have tested the algorithm without the feedback of acc or mag, and contrasted with the direct integration of the gyro data. The results show that the quaternion implementaion with a half gyro is more stable than the direct integration of single axis gyro data, which has less drift over time while the direct integration caused drift is very obvious.
Therefore, I'm wondering why the fusion algorithm deal with the gyro, acc, and mag data by scaling 0.5. Could you please give me some learning material regarding the quaternion implementation for devriving pose from gyro data in this code.
Many Thanks!

xioTechnologies, 2023-01-09:
Variables such as `halfGravity` and `halfGyroscope` exist as an optimisation to avoid unnecessary operations.  There any many intermediate calculations that involve a multiplication by 2.  The optimisation removes all of these intermediate multiplications, and replaces them with a single multiplication at the end.

For example, consider the following sequence:

1. `value = 2 * input`
2. `value = value + 2 * constant_a`
3. `value = value + 2 * constant_b`
4. `output = value`

This can be optimised as:
1. `half_value = input`
2. `half_value = value + constant_a`
3. `half_value= value + constant_b`
4. `output = 2 * value`

This is effectively what is going on in [FusionAhrsUpdate](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L100).

jacklee0575, 2023-01-10:
Thank you for your kind reply of the detailed trick in the code. I also want to ask the second question. I have tested the algorithm without the feedback of acc or mag, and contrasted with the direct integration of the gyro data. The results show that the quaternion implementaion with a half gyro is more stable than the direct integration of single axis gyro data, which has less drift over time while the direct integration caused drift is very obvious. Have you noticed this phenomenon? Could you tell me the possible reason for this phenomenon. Many thanks!

jacklee0575, 2023-01-11:
Thank you for your detailed reply. I have understand this phenomenon.

closed 2023-01-11 by jacklee0575