xioTechnologies/Fusion #65: Understanding FusionMatrix gyroscopeMisalignment

LazaroFilm, 2023-01-30:
When I rotate my sensor around the Z axis (on a turntable), I get the below graphs which shows some wobble in the roll and pitch with `.gain` to set to `0.0f` to ignore accel input.
I think I understand this "wobble" is due to a gyro axis misalignment. I'm trying to understand how the `FusionMatrix` works, but I'm not quite sure of the method to get the values needed for it. I understand each values is a multiplier to add some of the other axis input in another one, but I don't know how to tell which value and by how much... I've found [this article](https://www.analog.com/en/analog-dialogue/articles/mems-imu-gyroscope-alignment.html) but it hasn't quite helped me that much.
 
[image]

and zoomed in version to focus on pitch and roll: 

[image]

xioTechnologies, 2023-01-30:
We do not have any tutorials or examples for the calculating a misalignment matrix.  The [article](https://www.analog.com/en/analog-dialogue/articles/mems-imu-gyroscope-alignment.html) you linked looks like a good place to start.

On a separate note, I suggest you use the gyroscope offset correct algorithm as demonstrated on [line 24](https://github.com/xioTechnologies/Fusion/blob/main/Examples/Advanced/main.c#L24) and [line 51](https://github.com/xioTechnologies/Fusion/blob/main/Examples/Advanced/main.c#L51).  Adding these two lines of code will correct for the drift apparent in your plots.

herbertvuijk, 2023-02-24:
This guy made a calibration that is very similar to the misalignment matrix used by the xio Fusion algo. It's for an accelerometer though, but I guess the principles are the same.

By the way I found out by testing that you indeed have to do like a 9-orthogonal calibration for the accelerometer as offsets and gains are different in every position. A gyroscope though, seems to have the same offsets regardless of orientation. So I guess you don't really need a misalignment gyromatrix. Instead you can just substract a constant value that you have measured. 

Here an example of a calibration matrix (offset, gain, angles): 

https://github.com/BasicAirData/Clinometer/blob/master/app/src/main/java/eu/basicairdata/clinometer/CalibrationActivity.java

closed 2023-03-02 by xioTechnologies