xioTechnologies/Fusion #67: Axis Interconnection/Bleed

FugLong, 2023-02-02:
Hi all, I'm implementing fusion on an arduino 33 ble to create a head tracker for PC games.

I am currently encountering a problem where my yaw and roll axis work as they should, but my pitch axis affects yaw as well as pitch. I'm hoping someone more knowledgeable in fusion (and math generally) could point me to some possible causes of this behavior so I can look into them. 

I have already ensured that the weird orientation on the 33 ble's mag is being compensated for when updating the quaternion, but the pitch axis still effects yaw somehow. Any details, explanations, or other advice would be greatly appreciated. 

Here are some plots demonstrating the behavior:

https://imgur.com/a/FJZQ8XB

xioTechnologies, 2023-02-03:
The [code](https://github.com/Reefwing-Software/Reefwing-AHRS) you are using appears to be significantly different to Fusion.  It also appears to have obvious mathematical errors.  For example, [line 781](https://github.com/Reefwing-Software/Reefwing-AHRS/blob/main/src/ReefwingAHRS.cpp#L781) of [ReefwingAHRS.cpp](https://github.com/Reefwing-Software/Reefwing-AHRS/blob/main/src/ReefwingAHRS.cpp) is an invalid conversion from NWU to NED.

I suggest that you use Fusion as provided in this repository.  If you need Fusion to use NED then please use [this branch](https://github.com/xioTechnologies/Fusion/tree/axes-convention) which will soon be merged into master.

FugLong, 2023-02-03:
@xioTechnologies Thank you so much, I will switch to use that branch and let the Reefwing maintainer know about the conversion error they added

closed 2023-02-03 by FugLong