xioTechnologies/Fusion #68: FusionOffsetInitialise() never uses user defined gyroscopeOffset.

LazaroFilm, 2023-02-04:
In my previous issue ticket, you noticed I had some drift in my gyro at startup and you had mentioned to use `FusionOffsetInitialise()` to solve it. It actually was already on, but there is an issue with that function. when called , [Line 44 of FusionOffset.c](https://github.com/xioTechnologies/Fusion/blob/89295ab1ba145d5486e46b769ee82e11375f90e1/Fusion/FusionOffset.c#L44) zeroes out the `gyroscopeOffset` value I defined in my sketch. So the gyro always starts off calibration, then, after a couple of minutes it slowly drifts to the center. Deleting the line seems to solve the issue.

Line 44:
`    offset->gyroscopeOffset = FUSION_VECTOR_ZERO;`

xioTechnologies, 2023-02-04:
Please provide a link to the previous ticket you are referring to, and to your code using Fusion.

LazaroFilm, 2023-02-07:
Previous ticket mentioned: #65 
Here is the repo. It's still very much a work in progress. https://github.com/LazaroFilm/CyberLevel

xioTechnologies, 2023-02-08:
My understanding is that you have defined the variable `gyroscopeOffset` on [line 20](https://github.com/LazaroFilm/CyberLevel/blob/main/CyberLevel_Nano_RP_FF/CyberLevel_Fusion.ino#L20) and you believe that [line 44](https://github.com/xioTechnologies/Fusion/blob/89295ab1ba145d5486e46b769ee82e11375f90e1/Fusion/FusionOffset.c#L44) will overwrite this variable with the assignment `offset->gyroscopeOffset = FUSION_VECTOR_ZERO;`.  If this is the case then you are mistaken.  `gyroscopeOffset` and `offset->gyroscopeOffset` are not the same variable.

closed 2023-03-02 by xioTechnologies