xioTechnologies/Fusion #77: Bug - Commit "Use designated initialisers" breaks compilation in C++

ladislas, 2023-02-08:
Downloading the latest source from main, it seems that https://github.com/xioTechnologies/Fusion/commit/ec507511cff0e7499e38f54f0e4eaa5abeffa99e introduced a bug when working in C++.

[image]

xioTechnologies, 2023-02-08:
Please can you confirm which compiler you are using.  Please can you also show me how you include Fusion in your .cpp files.  You should only use `#include "Fusion.h` because this file wraps the library in `extern "C"`.

ladislas, 2023-02-08:
> Please can you confirm which compiler you are using

```
arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 10.3-2021.10) 10.3.1 20210824 (release)
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

> Please can you also show me how you include Fusion in your .cpp files

Here: https://github.com/leka/LekaOS/blob/ladislas/feature/replace-mahony-with-new-fusion/libs/IMUKit/source/IMUKit.cpp#L13

xioTechnologies, 2023-02-09:
We are investigating this.  In the mean time, please either use the [latest version](https://github.com/xioTechnologies/Fusion/releases/tag/v1.0.7), or revert https://github.com/xioTechnologies/Fusion/commit/ec507511cff0e7499e38f54f0e4eaa5abeffa99e to avoid the issue.

xioTechnologies, 2023-02-10:
The issue should be fixed as of https://github.com/xioTechnologies/Fusion/commit/dde922f72a717aa251ee1c18d242f57bb6bc2de2.  Please can you confirm the fix on your system.

ladislas, 2023-02-19:
I can confirm that the issue has been fixed, thanks a lot!

closed 2023-02-19 by ladislas