xioTechnologies/Fusion #82: Couldn't download imufusion Python package

limited-fion, 2023-02-10:
I couldn't install the python package. Was wondering if I missed anything..

Error message:
```
pip install imufusion
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement imufusion
ERROR: No matching distribution found for imufusion
```
OS Version:
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
Python version: 3.9.2
Numpy version: 1.24.2

xioTechnologies, 2023-02-10:
Please retrying after updating to the latest version of Python and Numpy.

limited-fion, 2023-02-11:
Python version updated to 3.10.0
numpy version is already
```
Package    Version
---------- -------
numpy      1.24.2
```

Still no luck :/

xioTechnologies, 2023-02-17:
The installation should now work.  Please can you confirm.

closed 2023-03-02 by xioTechnologies