xioTechnologies/Fusion #84: Accuracy and Precision Compared to the MadgwickAHRS

mecitpmk, 2023-02-15:
Hello Mr.Sebastian,

I have a some research about the library which you provide in this repository.

When I work with the old version of MadgwickAHRSupdateIMU which you provided at "19/02/2012 SOH Madgwick Magnetometer measurement is normalised" , I record the Pitch and Roll angles for a 60 seconds and i get the results as follows : 

Pitch angle error : 0.001990 Degree
Roll   angle error : 0.001998 Degree

However, when I implement same structure with Fusion repository and record them for a 60 seconds i get bad result than the first one. The results as follows for Fusion Lib. : 

Pitch angle error : -0.005449 Degree
Roll   angle error : 0.009339 Degree

Settings : gyroMisalignments , gyroscopeSensitivity , accelerometerMisalignment and accelerometerSensitivity was setted as default.
- Sensor sampling rate is 1000Hz
- Beta was setted as 0.02 in old version of MadgwickAHRS
- Fusion gain also setted as 0.02 in the test.

Calibration Process for both repo (MadgwickAHRS version and Fusion version) offsets calculated as follows : 
- Collect accel,gyro data for 5 seconds while the device at rest ( zero motion ) when the system has been started.
- Sum them up, and divide it into 5000 ( sample rate = 1000Hz, collect data for 5 sec  equals to 5000)
- Only Accel z axis offset = 1 - offsetValue

Therefore, although all the process are same with the MadgwickAHRS version the deviation in the pitch and roll angle is much higher.
By the way, I didn't forget to put FusionOffsetUpdate in every calculation process and I dont work with Magnetometer.

What is your suggestion in that case  how I can improve the accuracy ? 

Thank you,
Abdulmecid Pamuk

xioTechnologies, 2023-02-16:
The calibration process you have described is unlikely to achieve an accuracy greater than 1°.  I don't see how your system is capable of evaluating a difference in error of 0.007°.  Am I missing something?

mecitpmk, 2023-02-16:
The error which I described like "Pitch angle error : 0.001990 Degree" calculated like this ; 
Summed all the collected pitch angle and divided to the collected sample amount in .csv file.

I assumed while there is no motion, the mean of the pitch angle should be close to the 0.0000 or sth.

By the way, I applied 6th Order LPF into raw accelerometer data , and 2th Order for the Gyroscope. Thats why, my pitch and roll angle means smaller than 1°.

The value which we get is acceptable of the system requirements but, I just want to trying make the system even better and precise.

Thank you,
Abdulmecid Pamuk

xioTechnologies, 2023-02-16:
The difference in error you have reported far exceeds the capabilities of your system.  It would be like someone using a voltmeter with 1 mV accuracy, and averaging it to claim they have achieved 7 μV accuracy.  I hope you see the problem with that.  Either way, I don't think I can be of much more help here.

mecitpmk, 2023-02-16:
I see your statement much clear right now. Thank you for the extending our horizon with the clear examples. Before the closing issue Mr.Sebastian, I have one more question;

Is it possible to implement Q1.31 fixed-point format to the your old repos? Do I need extra scaling to prevent overflow (while calculating auxiliary variables and gradient descent algorithm)? I’m totally open your suggestion.

Best Regards,
Abdulmecis Pamuk

xioTechnologies, 2023-02-16:
I advise against a fixed-point implementation.  It would not be an easy task and it would introduce a significant risk of errors.  Furthermore, I doubt the efficiency would improve much beyond the current floating-point implementation, especially given that FPUs are now common in many microcontrollers.

mecitpmk, 2023-02-16:
Thank you for everything Mr.Sebastian. 

Now I’m closing the thread and  again thank you.

closed 2023-02-16 by mecitpmk