xioTechnologies/Fusion #85: X axis never change

srchang2023, 2023-02-16:
Hi there, 

First of all i would like to thank you for you work, it's simply quite amazing, second i tried to use the gait tracking script, but when use the  FusionAhrsGetEarthAcceleration(&ahrs); the x axis never change, indeed the z axis neither, but any way as an example here's what i'm getting as output,

First read:
earth.axis ={
x	float	1494086.62	
y	float	1.10672033e-007	
z	float	-0.99994278	
}

Now moving;
earth.axis ={
x	float	1494086.62	
y	float	4.33112182e-005	
z	float	-0.999985158	
}

As stated before, the same value for Z and X axis, so **yes**, the IMU is located at my foot, **yes** i made soft and hard movements, **yes** the Euler angles works, **yes** the library used is the last version of fusion, btw;

The MCU used is a STM32, i guess it doesn't matter, but what could be important is the IMU, is a lsm6dso from st, the settings are:

- acelerometer = 2g @ 1667Hz
- gyroscope = 2000 dps @ 6667Hz
- Sample rate = 2 

I'm not using the magnetometer, but as far as i can tell it is not used by the function.
So any light on this issue would be very appreciated,

xioTechnologies, 2023-02-16:
There are several problems with the information you have shared.
1. An acceleration value of 1,494,086.62 g is clearly incorrect, especially given your sensor range is 2 g.
2. An accelerometer range of 2g is too low for expected motion of a foot, I suggest you use the maximum range of your device.
3. A sample rate of 2 Hz  is too low.  I suggest you use at least 100 Hz, ideally higher.

srchang2023, 2023-02-16:
Thank you so much for you time, and for helping me of course,
Made the changes suggested and the output is still the same, i suspect there's something wrong with the sensor settings parameters,
The calibration method for the acelerometer was put the sensor on a flat surface to measure the raw value, i took for 1g a raw value of 16384 according to the [internet](https://www.i2cdevlib.com/forums/topic/4-understanding-raw-values-of-accelerometer-and-gyrometer/), so then i sum or subtract the offset and put'em inside the FusionVector accelerometerOffset array, and the same for the gyroscope, according to this [site](https://makersportal.com/blog/calibration-of-an-inertial-measurement-unit-imu-with-raspberry-pi-part-ii#:~:text=The%20simplest%20calibration%20of%20an,the%20IMU%20is%20not%20moving.) all i had to do is leave stan still the sensor and add the values to the gyroscopeOffset array, so in theory the sensor is calibrated, but as you  say, "_An acceleration value of 1,494,086.62 g is clearly incorrect, especially given your sensor range is 2 g._" then i'm missing something but i can not find out what is, so again any help would be very handy.

xioTechnologies, 2023-02-16:
I am sorry but I am unable to help.  I suggest you search for online tutorials and guides for how to read measurements from the sensors.

srchang2023, 2023-02-16:
I'm gonna start from it, what's weird is that Euler angles works, but anyways, thanks so much for you help, maybe i'll be back with more doubts later,
Regards!