xioTechnologies/Fusion #91: Get displacement point vector

kwe3hk, 2023-02-20:
I wonder if it is possible to obtain a point vector corresponding to the initial position when the motion started?

xioTechnologies, 2023-02-20:
The following is from the FAQ on the x-io website.

_Can I measure position using an IMU?_

_IMUs do not provide a direct measurement of linear velocity or position. IMUs provide inertial sensor (gyroscope and accelerometer) measurements alongside a measurement of 3D orientation relative to the Earth’s surface, often expressed as the Euler angles roll, pitch, and yaw._

_Our IMUs also include a magnetometer so that orientation is measured relative to magnetic north, and a sensor fusion algorithm to provide measurements of linear acceleration and earth acceleration. Earth acceleration is the [second derivative](https://en.wikipedia.org/wiki/Second_derivative) of the IMU’s position in the Earth coordinate frame._

_An IMU can be used to indirectly measure linear velocity and position in certain applications. However, this is only possible using specialised algorithms specific to the given application. For example, we have demonstrated how an [IMU attached to a person’s foot](https://github.com/xioTechnologies/Gait-Tracking) can be used to precisely track the 3D position of the wearer; or how [oscillatory motion](https://x-io.co.uk/oscillatory-motion-tracking-with-x-imu/) such as the heave of an ocean buoy or the respiratory motion of a patient’s chest during a medical scan can be measured using an IMU._

_It is important to understand that these demonstrations and algorithms are highly specialised and cannot be repurposed for generalised 3D tracking. Our IMUs are the ideal research tool for those wishing to develop tracking solutions using IMU technology. However, if you require an off-the-shelf, general-purpose, position tracking solution then you should consider alternative technologies such as GNSS or camera-based systems._

kwe3hk, 2023-02-20:
Thank you very much for your information :)

kwe3hk, 2023-02-20:
:)

closed 2023-02-20 by kwe3hk