xioTechnologies/Fusion #92: Automatic / dynamic gain calculation

herbertvuijk, 2023-02-24:
In my application the amount of noise varies a lot. 98% of the time there is a lot of noise in the data (a high standard deviation on every accelero and gyro measurement)  - Period HN, 2% of the time there is much less noise (a very low standard deviation) but there is high acceleration - Period LN. 

I need the most accurate orientation determination period LN, orientation during HN I don't care about. 

The problem is that if I take a gain of 0.5 to deal with HN, the orientation determination is not so good during LN. If I take a gain of 0.1  the orientation determination works well for LN but get pretty off during the HN. 

Although I'm not interested in orientation during period HN I also need it to be fairly decent, because otherwise I start of my calculations in the LN period based on a wrong start orientation. During the HN period the device is relative in the same orientation, but exposed to a lot of noise (bouncing).

What I ended up doing was toggling the Gain programmatically between 0.5 and 0.1 based on my own HN/LN detection. This works pretty reasonable, but is not so sophisticated. I learned that in some data 0.5 /0.1 works , in other 0.4/0.2, e.t.c. Also I'm nog happy with the hard toggling, I guess I should implement some ramping up/down. 

Is there any recommendation on how to find optimal and dynamic Gain values?

xioTechnologies, 2023-02-24:
Best performance is achieved with a fixed gain and well calibrated sensors.  I expect that the improvements achieved through a dynamic gain could be shown to be a workaround for some underlying issue that should be fixed.

Please can you describe your application in a little more detail.  What sensors you are using?  What calibration has been performed?  What is the motion being measured?  It would also be useful if you could share data demonstrating the issue, either as plots or as CSV files.

herbertvuijk, 2023-03-07:
I'm using gyro + accelero, calibrated as good as possible considering the equipment I have, deriving offsets, gains and angles from sampling on every axis. 

The use case is to calculate jumps (heights) during kiteboarding, www.thesurfr.app . People take a phone or a watch on the water and start jumping around. I detect the jump patterns and calculate the height of the jump by double integrating the earth vertical acceleration (z). Normally this double integration goes on for anywhere between 5 and 10 seconds.

Here a cool example of some action: https://www.instagram.com/reel/CoCbtwZoXFi/?utm_source=ig_web_copy_link where a person jumps 27 meter in the air tracked with our app. This version of the app is currently using an complementary based filter algo that I created myself, but I'm considering to move to the xioTech and/or Madgwick one. 

So a jump of a kiter starts at ground level (0 meter height) and I calculate the movement all the way through till the end of the jump. In a perfect world (with perfect orientation tracking, perfect sensors, ..) in theory my calculations should end at 0 by the end of the jump. In reality the double integration causes quite some drift and the calculation basically never ends at zero, but could be off by a large margin. Here you see such a jump, where the orange values are the results of double integrating the blue values (earth vertical acceleration). It's start with a high acceleration take-off, than floating town till the person hits the water. 

[image]

When the kiteboarder in the air, is the period I referred to as Low Noise before. Because the person is in the air, there is a lot of lineair acceleration and some rotation but there is not so much noise. As you can see in the image when the person is airborn the variation / std dev is relative low. 

The opposite is true when the person is on the water. Then there is a lot of noise and not so much lineair acceleration or rotational movements. Here a snapshot on how the earth acceleration looks when the user is on the water:

[image]

I ended up with a situation where I use a gain of 0.5 while the person is on the water, but as soon as the person takes off I use a gain of 0.1. Because a gain of 0.5 while air born throws the orientation completely off*. The 0.1 gain is also not ideal, sometimes other values work better, depending on how solid the user attached the phone or watch. I learned that even way people attach their wrist band (in case of a watch) can call for another ideal gain. Which is why I'm looking for some way to automatically calculate / determine the optimal gain based on observed values/drift.

*meaning that the double integrated position calculations drifts even more. Here an example of the same jump as the first picture but now with a 0.5 gain, as you can see my double integrated calculation drifts even more away from  0   (where it should be at the end in a perfect world), in this case it drifts to -10. To me this drift is an indication of worse orientation tracking. 
[image]

Any tips specific to my use case on how to potentially improve?

LazaroFilm, 2023-03-08:
You may want to look into using two accelerometers one high G and one low g like the [X-IMU3](https://x-io.co.uk/x-imu3/) which may be a good device to use in your case, you may want to add one extra layer of waterproofing to it though.

xioTechnologies, 2023-03-08:
Thank you for explaining.  

I understand how switching between two gains benefits your application but this behaviour and the associated heuristics are extremely application-specific.  The dynamic gain you have described would not be an appropriate feature to introduce to Fusion.  Fusion does not prevent you from implementing this behaviour yourself.

You said that you ramping the gain up/down may be preferable.  I do not see any benefit in this over hard toggling.

xioTechnologies, 2023-03-08:
I understand how switching between two gains benefits your application but this behaviour and the associated heuristics are extremely application-specific.  The dynamic gain you have described would not be an appropriate feature to introduce to Fusion.  Fusion does not prevent you from implementing this behaviour yourself.

You said that you ramping the gain up/down may be preferable.  I do not see any benefit in this over hard toggling.

herbertvuijk, 2023-03-08:
Thanks all.. @LazaroFilm the idea is that people use existing hardware (phones, watches) and therefor its not really possible to use an external IMO. 

@xioTechnologies Thanks you, if I discover anything interesting with regards to dynamic gain I will let you know.

closed 2023-03-08 by herbertvuijk