xioTechnologies/Fusion #94: integration in c++ projects

sfbence, 2023-03-14:
Absolute great library, have been using in python and outperformed all other alternatives. Nevertheless still because of performance issues I an switching to use it in c++. Although the code compiles perfectly as C, I am having hard time compiling the C++ code using this header. Could you give an example how it was meant to compile (gcc or g++) for c++ projects? Thank you in advance!

xioTechnologies, 2023-03-15:
We have not included a C++ example because it would be identical to the C example.  Please can you be more specific about the issue.  If you are seeing a compiler error then please report that error along with other information that may be useful for us to look into the problem.

LazaroFilm, 2023-03-15:
Have you pulled the most recent version? There used to be an issue with C++ compiling that was fix d a while ago.

sfbence, 2023-03-15:
Thanks, apparently I had still the old version, now it works!

closed 2023-03-15 by sfbence