xioTechnologies/Fusion #95: accelerometer error addition during the AHRS updating step.

sinitsky96, 2023-03-23:
Hello,
After reading the thesis and going over the code of the revised AHRS, I've noticed the error of a sensor is calculated and stored in the Internal state but i could not find its usage during the updating step.
Would very much appreciate the help.
Thank you.

xioTechnologies, 2023-03-23:
The error in inclination and heading is used to calculate the feedback on lines [143](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L143) and [172](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L172) respectively.  These feedback terms are then applied on line [188](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionAhrs.c#L188).

sinitsky96, 2023-03-24:
Thank you very much.

closed 2023-03-24 by sinitsky96