xioTechnologies/Fusion #96: Questions about the scope of application

WYW0, 2023-04-06:
Hello！
I collected data on a set of circular trajectories in the horizontal direction using sensors. After using these data to perform fusion algorithm calculations, strange quaternions were generated. I want to know if the fusion algorithm is not good at this method. Thank you.
[image]

xioTechnologies, 2023-04-06:
You appear to have extremely high gyroscope offset errors.  I suggest that you verify that when you device is stationary, the gyroscope measures 0 degrees/s in all three axes.

WYW0, 2023-04-06:
I re collected data with the same trajectory but more stable, and achieved acceptable results. And I noticed that x-IMU3 is about to be released. Can you tell me its overall dimensions?
[image]

xioTechnologies, 2023-04-06:
I am pleased you've achieved acceptable performance.  Please close this issue if it is now solved.  x-IMU3 information is provided on the [x-IMU3 webpage](https://x-io.co.uk/x-imu3/).  Please contact x-io via the website for product enquires.

closed 2023-04-07 by WYW0