xioTechnologies/Fusion #97: I have a question about your Fusion algorithm in C code

JiyoonJo, 2023-04-06:
When rotating the sensor, roll, pitch, and yaw are all normally calculated as Euler angles. However, when the sensor stops rotating, only the yaw value does not stop and continues to change to the initial state value.

Can you tell me why it's like this?

xioTechnologies, 2023-04-06:
Please can you share a plot to better explain the issue.  For example, like the questioner in [this issue](https://github.com/xioTechnologies/Fusion/issues/96) has done.

JiyoonJo, 2023-04-07:
I can't attach a image file of graph becasue of the security policy for my company.
Can i send a image by e-mail?
Let me know your e-mail address.

thanks for your help.

xioTechnologies, 2023-04-07:
Please contact us using the [website](https://x-io.co.uk/contact).

closed 2023-04-07 by xioTechnologies