xioTechnologies/Fusion #103: [Assumption] That code does not work properly

Overff, 2023-05-20:
Hello.
I spent a lot of time to setup and run up that code. Unfortunately with no success.
My goal is to calculate heading by using MAG and ACC. I used FusionCalibrationMagnetic() in that purpose. 

In order to calibrate MAG and ACC i tied to use two different ways: 1. MagMaster app and 2. Magneto. These applications generate transformation matrix (soft iron) and bias (hard iron offset).  Then I used outputs from these applications in your code (check out **Annex1**)
So
I assume that FusionCalibrationMagnetic() does not work properly with your calibration model (mc = Smu - h). Fortunately, if use regular model (mc = S(mu - h)) it works better but actually not stable.
 
_mc is the calibrated magnetometer measurement and return value
mu is the uncalibrated magnetometer measurement and uncalibrated argument
S is the soft iron matrix and softIronMatrix argument
h is the hard iron offset vector and hardIronOffset argument_

So, I assumed that code does not work properly (include yaw, roll, pitch). Visualization by using processing confirm my assumption. 
My setup: teensy board and different IMUs(9250, Gy86, Gy91 and etc)

It would be great to see/read applicable instruction to how it use. 

**Annex1** (example of my code with soft and hard iron):

_//Accelerometer transformation matrix  
    const FusionMatrix accelerometerMisalignment = {0.997900, -0.013244, -0.000926, -0.013244, 0.998071, -0.002810, -0.000926, -0.002810, 0.972633};
    const FusionVector accelerometerSensitivity = {1.0, 1.0, 1.0};
    const FusionVector accelerometerOffset = {-0.001079, 0.009427, -0.020567}; //Bias
       
    //Magnetometer transformation matrix
    
    const FusionMatrix softIronMatrix = {0.076813, 0.000005, -0.000037, 0.000005, 0.073755, -0.002360, -0.000037, -0.002360, 0.084221};
    const FusionVector hardIronOffset = {64.808407, 277.759814, -536.878619}; //Bias_

xioTechnologies, 2023-05-20:
The function `FusionCalibrationMagnetic` implements the expression m<sub>c</sub> = Sm<sub>u</sub> - h, as described in the [README](https://github.com/xioTechnologies/Fusion#readme).  This implementation in code is as follows (see [line 37](https://github.com/xioTechnologies/Fusion/blob/main/Fusion/FusionCalibration.h#L37)).
```
static inline FusionVector FusionCalibrationMagnetic(const FusionVector uncalibrated, const FusionMatrix softIronMatrix, const FusionVector hardIronOffset) {
    return FusionVectorSubtract(FusionMatrixMultiplyVector(softIronMatrix, uncalibrated), hardIronOffset);
}
```
I do not see any issue with the code.  If you believe that the code does not work then please demonstrate this.

PetSven, 2023-09-14:
I am also wondering if FusionCalibrationMagnetic is wrong. Matlab uses the equation: 
[image]
Vectornav uses the equation: 
[image]

xioTechnologies, 2023-09-15:
Both $m = Su - h$ and $m = S(u - h)$ are valid but $m = S(u - h)$ does seem to be in more widespread use.  The choice of $m = Su - h$ in Fusion is not arbitrary.  Decoupling $S$ and $h$ provides several practical advantages.

It is also my understanding that $m = Su - h$ better represents the physical phenomena concerned.  For example, if we were to start with a calibrated magnetometer providing its measurement $m$, and then attach a magnet which had an intensity of $n$ at the location of the magnetometer, then we would have introduced a new hard-iron offset and the measurement $m$ would become $m + n$.  If we wanted to update our calibration to correct for this then we would replace $h$ with $h + n$.  In this case, $h$ directly represents the physical hard-iron offset.

If we were to instead use $m = S(u - h)$ then the correction would require we replace $h$ with $h + S^{-1}n$.  In this case, $h$ does not directly represent the physical hard-iron offset.

That said, conventions and standards are important.  If there is a general consensus for $m = S(u - h)$ then it may render arguments for the alternative moot.  I would be interested to hear other's thoughts on this.

xioTechnologies, 2023-09-25:
The magnetic calibration model has been updated to $m=S(u-h)$ in the latest release, [v1.2.0](https://github.com/xioTechnologies/Fusion/releases/tag/v1.2.0).

closed 2023-09-25 by xioTechnologies

Overff, 2024-01-05:
Thy for update!

AndreyDiDev, 2024-06-10:
@Overff how did you get the accelerometer transformation matrices? I couldn't find MagMaster, but both seem for the Magnetometer ?