xioTechnologies/Fusion #105: I have a question about the init state

JiyoonJo, 2023-06-01:
When operating the algorithm filter for the first time, the values of pitch, roll, and yaw will have different values in the beginning depending on the slope of the sensor, should I change the init value below? (q_est = { 1, 0, 0, 0})

struct quaternion q_est = { 1, 0, 0, 0};       // initialize with as unit vector with real component  = 1

The figure below shows the state in which the imu is installed at a different angle.

1) Standing on the side

  ________________  
 |                        | 
 |  *                    | 
 |                        | 
 |   imu sensor   | 
 |                        | 
 |                        | 
 |________________|

2) Layed down
  __*_____________  
 |__imu_sensor___| 
 

3) Reverse and lie down 
  ________________  
 |___imu_sensor__| 
    *

xioTechnologies, 2023-06-01:
There is no variable `q_est`.  Please use the latest code available in this repository.  The algorithm already handles initialisation, as [described](https://github.com/xioTechnologies/Fusion#initialisation) in the README.

JiyoonJo, 2023-06-01:
ok thank you for your help.
I will try to apply latest code.

closed 2023-06-06 by JiyoonJo