xioTechnologies/Fusion #107: earth acceleration direction convention_enu

songtao-git, 2023-06-23:
Hello, I would like to confirm if my understanding of the direction of earth acceleration is correct. If I specify CONVENTION_ENU, then x corresponds to the north, y corresponds to the east, and z corresponds to downwards?

I am setting different heading values in ahrs. Will this affect x， y and z of earth acceleration?

The complete scenario for which I am consulting this question is for a car-mounted inertial navigation system. I can obtain position, heading and velocity data through GNSS:
1. Set the initial position and velocity through GNSS data, including heading if necessary.
2. Calculate the vehicle’s acceleration through ahrs’ earth acceleration, then speed, and then new position. This can achieve the scenario of car-mounted inertial navigation.
3. If there is a need for calibration during parking or uniform straight-line motion, GNSS data can be used to calibrate the inertial navigation system.

I am eagerly waiting for your reply

xioTechnologies, 2023-06-23:
ENU means East = X, North = Y, and Up = Z.  The three letters correspond to XYZ, in order.  The heading will determine how accelerometer measurements map onto the Earth's XY plane.

songtao-git, 2023-06-25:
Thanks！Is the heading angle relative to North? In ENU and NED, does the same heading value mean the same thing?

xioTechnologies, 2023-06-25:
[Axes Conventions](https://en.wikipedia.org/wiki/Axes_conventions) describe how the XYZ directions are assigned to the Earth coordinate frame.  Fusion supports three different axes conventions: North-West-Up (NWU), East-North-Up (ENU), and North-East-Down (NED).  If you are confused then I suggest you either ask a specific question, or search online to fill the gaps in your knowledge.

closed 2023-06-26 by songtao-git