xioTechnologies/Fusion #109: How to use imufusion's python package to calculate the trajectory through nine-axis data? Could you give an example? Thanks a lot.

One-one-one-learner, 2023-06-28:


xioTechnologies, 2023-06-28:
It is typically not possible to calculate a 3D trajectory using only IMU measurements.  I suggest you read the _"Can I measure position using an IMU?"_ FAQ on [this page](https://x-io.co.uk/contact/).

closed 2023-06-28 by xioTechnologies

One-one-one-learner, 2023-06-29:
> It is typically not possible to calculate a 3D trajectory using only IMU measurements. I suggest you read the _"Can I measure position using an IMU?"_ FAQ on [this page](https://x-io.co.uk/contact/).

Thank you for your reply. 

After reading the answer, I found that the IMU can give the motion trajectory under certain circumstances, such as gait tracking and [oscillatory motion](https://x-io.co.uk/oscillatory-motion-tracking-with-x-imu/), so why can't the trajectory tracking be realized only under the condition of IMU?  Could you give some hints or reasons to explain? Looking forward to your reply.

xioTechnologies, 2023-06-29:
Because of the drift associated with double integration.  You will find much better explanations and discussions of this if you search online.