xioTechnologies/Fusion #118: Dynamic yaw angle calculation.

dihaoAI, 2023-08-15:
Hi，
thanks for the amazing algorithm.  We have achieved a satisfied result with ‘update_no_magnetometer’ function and "acc+gyro" data in our device for a common scene. We get the right yaw angle with less drift. However, we have a specific application to calculate yaw angle. As shown in the figure, when we rotate the device with z-axis, it also rotates with x-axis (approximately 3-15dps). And we want to get the yaw angle when the device is stationary. In other words, we did not calculate any angle when gx>T(approximately 3-15dps).
Would you give some hints or suggestions? We tried a few approaches but not work, for example, set the sensitivity=[0, 0, 1] for gyro calibration.

Thanks a lot.
[image]

dihaoAI, 2023-08-15:
Thanks for the quick reply.

We want to get the change of yaw angle using ‘update_no_magnetometer’ function and "acc+gyro" data. The picture above is a driller with our device. As we rotate the driller around the z-axis, the driller is also rotating around the x-axis.  For example, the initial yaw of driller is 0, then the driller turns on and starts to rotate around the x-axis. At this time, we move the driller 30 degrees around the z-axis and turns off the driller. After the driller turns off, it lies in a state of stasis. And we expect to get the yaw angle of the driller is 30 degree.

xioTechnologies, 2023-08-15:
Your description is still unclear.  My understanding is that you mounting a low-cost, uncalibrated IMU on a spinning drill bit, and expecting to obtain a reliable measurement of heading.  I suggest you rethink your approach entirely.  The scope is beyond what can be discussed here.

dihaoAI, 2023-08-15:
Thanks a lot.

Yes, I am using a mpu6500 on a drill bit and want to get the relative change of heading. When the drill switch is not activated (the value of $gx$ approximately equal to 0), I can get the right result. When the drill switch is on (the value of $gx$ approximate equal to 100dps or greater), the heading value is incorrect. 
Another point is that we don't need to measure the heading while the dill is turned on.

xioTechnologies, 2023-08-15:
I suggest you mount the IMU on the drill main body, not the drill bit.  If you require an IMU on the drill bit then I suggest you use two IMUs, one on the drill bit and the other on drill main body.  You can use the heading measurement from the drill body IMU to set the heading of the drill bit IMU using `FusionAhrsSetHeading`.

dihaoAI, 2023-08-16:
hi，thanks for your suggestion and sorry for the confusing expressions. Actually,  I have used two IMUs as you said and I need to calculate the yaw of drill bit. I provided two csv date file to hopefully better describe my problem. 

1. In **'rotate-drill off.csv'**, I rotated the drill around z-axis and back to the origin. In this process, the dirll is off.
2. In **'rotate-drill on.csv'**, I also rotated the drill around z-axis and back to the origin. In this process, the dirll is on.
[rotate-drill off.csv](https://github.com/xioTechnologies/Fusion/files/12354838/rotate-drill.off.csv)
[rotate-drill on.csv](https://github.com/xioTechnologies/Fusion/files/12354839/rotate-drill.on.csv)

The fig1 is the **simple_example.py** output result for **'rotate-drill off.csv'**. It seems I got the right results.
[image]

The fig2 is the **simple_example.py** output result for  **'rotate-drill on.csv'**.  I used **FusionCalibrationInertial** and the sensitivity is  set **[1, 1, 1]**.
[image]

The fig3 is the **simple_example.py** output result for  **'rotate-drill on.csv'**.  I used **FusionCalibrationInertial** and the sensitivity is  set **[0, 0, 1]**.
[image]

It seems the fig3 also give the right yaw angle. However, the result of yaw angle is opposite to the fig1. Would you give me some help? Thanks a lot.

xioTechnologies, 2023-08-16:
The sensitivity vector is 3 elements corresponding to the X, Y, and Z axis.  A sensitivity of 1 for each axis would therefore be `[1, 1, 1]`.  I do not understand why you would have `[0, 0, 1]`.  You will not be able to obtain a reliable measurement of heading from an IMU mounted on the drill bit.  You should use the IMU mounted on the drill body to obtain the measurement of heading.

dihaoAI, 2023-08-16:
Thank for your reply again. 
As you said before, I have two IMUs on the drill, one is on the drill body and the other is on the drill bit. In practice application, the drill bit will move around z-aixs in some specific scenarios. The main aim of the device is to monitor that if there is a heading bias between the drill body and drill bit. Therefore, I want to measure the heading angle of drill bit.

Thanks a lot.

closed 2023-08-21 by dihaoAI