xioTechnologies/Fusion #119: Python Implimentatoin

gbakerLV, 2023-09-24:
I am some what new to this.
I am getting an error trying to run in Python  
I installed imufusion, and running advanced_example.py in VSCode.

VS Code is having trouble resolving, ahrs_internal_states.acceleration_recovery_trigger and acceleration_recovery (there maybe others).  I can follow what you are doing to translate the calls from C to Python and I don't see any discrepancies.  

AttributeError: 'imufusion.InternalStates' object has no attribute 'acceleration_recovery_trigger'. Did you mean: 'acceleration_rejection_timer'?

If you could be so kind, to take a quick look.  thx

xioTechnologies, 2023-09-24:
The problem is that the main branch is currently ahead of the latest release.  You need to use the version of `advanced_example.py` from the [latest release, v1.1.2](https://github.com/xioTechnologies/Fusion/releases/tag/v1.1.2).

xioTechnologies, 2023-09-25:
The main branch now matches the latest release, [v1.2.0](https://github.com/xioTechnologies/Fusion/releases/tag/v1.2.0).

closed 2023-09-25 by xioTechnologies