xioTechnologies/Fusion #124: Allow configuring XYZ order in FusionQuaternionToEuler

wheaney, 2023-09-26:
I found that what I consider "roll" was being applied second in Euler ordering when passing through the `FusionQuaternionToEuler` function, making for really funky gimbal-locking type behavior when rotating my device +/- 90 degrees. I had to actually create a copy of the function in my own repo and rewrite it to get the ordering I needed. The gimbal locking would still be there, but since it's moved to a different axis, it's not really going to be hit in my application after moving it. It would be nice if this function allowed for choosing the ordering.

xioTechnologies, 2023-09-26:
XYZ and ZYX Euler angles are both subject to gimbal lock when the rotation around Y approaches ±90°.  Use of XYZ Euler angles would therefore not achieve your solution you are suggesting.  Furthermore, XYZ Euler angles are unconventional.  Most applications use either [ZXZ or ZYX](https://en.wikipedia.org/wiki/Euler_angles#:~:text=ZXZ%20for%20proper%20Euler%20angles%20and%20ZYX%20for%20Tait%E2%80%93Bryan).  Are you able to share a link to demonstrate XYZ Euler angles being used, or a specific requirement for this sequence?

I suggest you avoid using Euler angles altogether and instead use either quaternions or rotation matrices to achieve the transformations and calculations required by your application.

wheaney, 2023-09-27:
[Here](https://github.com/wheaney/xrealAirLinuxDriver/blob/sombrero/src/driver.c#L255-L266) is the rewrite of the function as I described it.

And [here](https://gitlab.com/TheJackiMonster/nrealAirLinuxDriver/-/blob/main/interface_lib/src/device3.c?ref_type=heads#L853-861) is the probably more relevant code that uses the Fusion library to get the quaternion orientation of the device.

xioTechnologies, 2023-09-27:
You have implemented YZX Euler angles.  This sequence will be subject to gimbal lock when the rotation around Z approaches ±90°.  Is this not an issue for your application?

I believe that [your description](https://github.com/wheaney/xrealAirLinuxDriver/blob/sombrero/src/driver.c#L255C4-L255C41), "_Fusion switches pitch and roll for us..._" is incorrect.  My understanding is that the angles are not switched in the sequence, they are just labelled differently.  There are many different naming conventions but this doesn't change the underlying mathematics.  An X rotation may be labelled _pitch_, _roll_, _elevation_, _inclination_, or _bank_ depending on the application and axis assignments.  If your application labels it _pitch_ and Fusion labels it _roll_, then you only need to change the labels, not the maths. 

I am currently unconvinced of the requirement to support non-standard Euler angle sequences.  I expect that your current solution is a workaround and that the proper solution is to use a quaternion or rotation matrix representation.

closed 2023-11-14 by wheaney