xioTechnologies/Fusion #128: Absolute Euler Threshold with Quaternions

Paddy8, 2023-09-29:
Hi, for my application I want to detect, if an angle crosses a specific threshold. Since it is not unusual for my system to reach pitch < -75° I encounter a gimbal lock quite often. Is it possible to define such thresholds for each angle by using just quaternions?

Another possibility I am thinking of would be by using a different rotation order than ZYX. Roll angle would be less problematic with gimbal lock.

Hope you can help me!

Best regards,

Patrick

xioTechnologies, 2023-09-29:
Fusion labels rotations around X, Y and Z as roll, pitch and yaw respectively.  The range of each Euler angle is:

- Roll = ±180°
- Pitch = ±90°
- Yaw = ±180°

If you wish to use Euler angles and your application expects pitch angles to approach ±90° then I recommend that you rotate your IMU so that these problematic rotations instead occur in the roll axis.  You can do this by physically rotating the IMU, or by virtually rotating the IMU using the following code

```
FusionVector newGyroscope = FusionAxesSwap(gyroscope, FusionAxesAlignmentPYNXPZ);
FusionVector newAccelerometer= FusionAxesSwap(accelerometer, FusionAxesAlignmentPYNXPZ);
```

It is possible to check if an orientation exceeds specified bounds without Euler angles but you would need to redefine your condition in appropriate terms.  You have described your condition as "_if an angle crosses a specific threshold_" which by definition, must use an angular representation.  Please can you describe your goal in boarder language.

Paddy8, 2023-09-29:
Thank you for your fast answer!
The idea is for example to detect when roll is <115 || >-115. To achieve this it might work to rotate the IMU virtually, like you mentioned above. I will for sure try this.
But it would be still interesting, how I would need to redefine my conditions? Because in the best case scenario I can check the orientation for exceeding specific bounds without bothering about gimbal lock.

xioTechnologies, 2023-09-29:
Please can you describe your application and the problem you are trying to solve.

Paddy8, 2023-10-03:
I can´t got into details because of confidentiality. But the idea is to track the movement of the human body and detect certain scenarios. The IMU is placed on the center of the back. 
For example, if a person is standing upright and bending forward its upper body, we come close to a pitch of 90°. Since such scenarios are happen often, we can´t really use Euler Angles to create thresholds for detecting events. 

So best case would be, if we can express all our boundaries/thresholds just with quaternions. To give you an maybe easy example. Let´s assume that a person will fall sideways if the IMU on the back experiences roll <120° || roll > -120°. How can we check this threshold all the time? Because if the person just bends forward far enough, we will encounter a gimbal lock and can´t trust the roll values anymore. 

Thank you for you help and sorry for the late reply!

xioTechnologies, 2023-10-03:
You have described that for an IMU attached to the back, the roll angle would represent sideways motion.  This is only true if the person is upright.  If the person bows forwards 90° then the roll axis aligns with the vertical so that the roll angle now represents the heading that the person is bowing towards.  The roll has become orthogonal to sideways motion.

Your statement, "_a person will fall sideways if the IMU on the back experiences roll <120° || roll > -120°_" is therefore incorrect.  I suggest you try and conceptualise the problem without invoking Euler angles.

closed 2023-10-04 by Paddy8