xioTechnologies/Fusion #129: Python 3.12 Install Issue

votecoffee, 2023-10-04:
I'm trying to install imufusion v1.2.0 with a python 3.12 venv.  I've already installed numpy v1.26.0.  I get the error below:
I know python 3.12 isn't officially supported yet for this package.  Any chance it could be updated?

> Collecting imufusion==1.2.0
>   Using cached imufusion-1.2.0.tar.gz (20 kB)
>   Installing build dependencies: started
>   Installing build dependencies: finished with status 'done'
>   Getting requirements to build wheel: started
>   Getting requirements to build wheel: finished with status 'error'
> 
>   error: subprocess-exited-with-error
>   
>   Getting requirements to build wheel did not run successfully.
>   exit code: 1
>   
>   [20 lines of output]
>   Traceback (most recent call last):
>     File "D:\~LocalWork\repos\IMUPathVibration\venv\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
>       main()
>     File "D:\~LocalWork\repos\IMUPathVibration\venv\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
>       json_out['return_val'] = hook(**hook_input['kwargs'])
>                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>     File "D:\~LocalWork\repos\IMUPathVibration\venv\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 118, in get_requires_for_build_wheel
>       return hook(config_settings)
>              ^^^^^^^^^^^^^^^^^^^^^
>     File "C:\Users\votec\AppData\Local\Temp\pip-build-env-ffto6nbq\overlay\Lib\site-packages\setuptools\build_meta.py", line 355, in get_requires_for_build_wheel
>       return self._get_build_requires(config_settings, requirements=['wheel'])
>              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>     File "C:\Users\votec\AppData\Local\Temp\pip-build-env-ffto6nbq\overlay\Lib\site-packages\setuptools\build_meta.py", line 325, in _get_build_requires
>       self.run_setup()
>     File "C:\Users\votec\AppData\Local\Temp\pip-build-env-ffto6nbq\overlay\Lib\site-packages\setuptools\build_meta.py", line 507, in run_setup
>       super(_BuildMetaLegacyBackend, self).run_setup(setup_script=setup_script)
>     File "C:\Users\votec\AppData\Local\Temp\pip-build-env-ffto6nbq\overlay\Lib\site-packages\setuptools\build_meta.py", line 341, in run_setup
>       exec(code, locals())
>     File "<string>", line 1, in <module>
>   ModuleNotFoundError: No module named 'numpy'
>   [end of output]
>   
>   note: This error originates from a subprocess, and is likely not a problem with pip.
> error: subprocess-exited-with-error
> 
> Getting requirements to build wheel did not run successfully.
> exit code: 1
> 
> See above for output.
> 
> note: This error originates from a subprocess, and is likely not a problem with pip.

votecoffee, 2023-10-04:
I added a couple of packages as per below:
```
pip install pandas
pip install wheel
```

I now get this error:
```
Collecting imufusion
  Using cached imufusion-1.2.0.tar.gz (20 kB)
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: imufusion
  Building wheel for imufusion (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [5 lines of output]
      running bdist_wheel
      running build
      running build_ext
      building 'imufusion' extension
      error: Microsoft Visual C++ 14.0 or greater is required. Get it with "Microsoft C++ Build Tools": https://visualstudio.microsoft.com/visual-cpp-build-tools/
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for imufusion
  Running setup.py clean for imufusion
Failed to build imufusion
ERROR: Could not build wheels for imufusion, which is required to install pyproject.toml-based projects
```

I'll install c++ build tools and try again, but it looks like adding pandas or wheel is needed.

votecoffee, 2023-10-04:
After installing MSVC c++ 2022 build tools, it installs fine.  Note that as per above, I had to pip install pandas and/or wheels first.  Feel free to close this if you want, but I'm leaving it open in case you want to add pandas and/or wheels to your dependencies.

xioTechnologies, 2023-10-05:
Fusion [v1.2.1](https://github.com/xioTechnologies/Fusion/releases/tag/v1.2.1) has just been released and supports Python 3.12.

Thank you for sharing your notes.  I found was not necessary to add _pandas_.  The only build dependencies are _numpy_ and _twine_, as per lines [77 and 78 of main.yml](https://github.com/xioTechnologies/Fusion/blob/main/.github/workflows/main.yml#L77) which is [running without issue.](https://github.com/xioTechnologies/Fusion/actions/runs/6415614693)

closed 2023-10-05 by xioTechnologies