xioTechnologies/Fusion #134: Rust bindings

avsaase, 2023-12-05:
Hi, thanks for your work. I would like to use this algorithm in a Rust project. I considered porting the code but then I found [this comment](https://github.com/jmagnuson/ahrs-rs/issues/21#issuecomment-1369023131) where you stated you'd prefer wrapping the C library. I don't know C all that well but I would like to give this a try. Would a Rust library be in scope for this repository?

mark2b, 2023-12-05:
I've ported this library to Rust and just published it hour ago
Please look at
https://github.com/mark2b/imu-fusion-rs
It is not binding, but Rust native implementation

xioTechnologies, 2023-12-05:
It sounds like @mark2b has what you need.

It is within the scope of this repository to provide a Rust library that wraps the C library, but we are unlikely to make progress on this until required by a project or client.

closed 2024-04-22 by xioTechnologies

avsaase, 2024-07-02:
HI @xioTechnologies, are you interested in a PR that adds Rust bindings? I gave it a try and it turned out to be quite easy.

xioTechnologies, 2024-07-03:
I would be interested in seeing the code but we would not merge external PRs.

avsaase, 2024-07-03:
Why is that?

You can find the code in my fork here: https://github.com/avsaase/Fusion/tree/rust-bindings

xioTechnologies, 2024-07-03:
To maintain quality.  Thank you for sharing.

avsaase, 2024-07-03:
Note that the bindings are currently not tested and could break in potentially subtle ways when the the C library changes.

I'll set up a separate repo and continue the development there.