xioTechnologies/Fusion #135: Guidance on tuning Fusion for "responsiveness"

copumpkin, 2023-12-22:
I'm not sure this is the right place to ask, but I'm wondering whether Fusion is well suited to detecting "snap" rotational movements of the IMU. Imagine that the IMU is traveling one direction, and then flips ~180º to travel at roughly the same rate in the other direction within less than a second. I expect Fusion's integration of the gyroscope and magnetometer data would detect this, but I'm not sure if the various smoothing and other compensation mechanisms built into the algorithm will delay detection of such events.

If so, do you have any advice on which parameters to tune in order to make the earth-relative orientation as responsive as possible? Ideally I'd like to be able to distinguish between a "snap" rotation like I described above and a longer smoother rotation, which could also arise in my domain. I imagine that there's a trade-off giving me higher noise for tuning the algorithm like this, but that's probably fine for what I'm doing with it.

Thank you!

xioTechnologies, 2023-12-23:
There is no issue in using Fusion for what you have described. The algorithm does not implement any smoothing or bandwidth-limiting processes.  There is no limit to the speed of orientation changes you can measure provided that your sensor measurement ranges and sample rates are high enough.

copumpkin, 2023-12-23:
That sounds great, thank you!

closed 2023-12-23 by copumpkin