xioTechnologies/Fusion #136: Inferring gyroscope range

copumpkin, 2023-12-23:
Your readme says:

> Gyroscope range (in degrees pe second). Angular rate recovery will activate if the gyroscope measurement exceeds 98% of this value. A value of zero will disable this feature. The value should be set to the range specified in the gyroscope datasheet.

(nit: typo in "degrees _pe_ second")

I'm using Fusion on IMU data from some hardware I don't have the datasheet for. Do you have any advice for how to figure out this parameter if I don't know it ahead of time?

copumpkin, 2023-12-23:
Here, I figured I'd just fix the typo: https://github.com/xioTechnologies/Fusion/pull/137

xioTechnologies, 2023-12-24:
Thank you for pointing out the typo.  If you are using an off-the-shelf product then I suggest you search for the product documentation online.  If you are using a custom-designed product then I suggest you ask the designer.  The `gyroscopeRange` is an optional setting that is only used to detect excessive angular rates.  If you are unsure then I suggest you leave the feature disabled (value of zero).

copumpkin, 2023-12-28:
Makes sense, thanks!

closed 2023-12-28 by copumpkin